XzuluX/EtherCAT.NET #41: Invalid watchdog configuration

mschwerin, 2020-02-28:
Hi,

I've got a slave based on an XMC4300 and SSC which I can successfully run to OP using both TwinCAT and the SOEM simple_test.

When I run your SampleMaster switching from SAFE_OP to OP fails. If I connect to the slave again after this using the SOEM simple_test I get the error "Invalid watchdog configuration".

Apparently your SampleMaster clears the watchdog register 0x0400 (ESC_WD_DIVIDER_OFFSET), and 0x0420 (ESC_PD_WD_TIME) somewhere / somehow. I can confirm this by debugging into my firmware. The registers return zero.

I tried writing the register myself but this does not work... Any help would be appreciated.

Thanks, Maximilian

Apollo3zehn, 2020-02-29:
Hi, the register 0x420 is cleared here: https://github.com/Apollo3zehn/EtherCAT.NET/blob/master/native/SOEM_wrapper/soem_wrapper.c#L559

Clearing the watchdog timer was necessary because C# is not real-time capable and a long I/O update delay due to garbage collection led to slave errors in my setups. I will try to publish an update tomorrow or on Monday to make the timeout configurable in the master.

mschwerin, 2020-03-01:
Hi,

thanks for the quick reply. I don't really need the watchdog to work but clearing register 0x0420 is not enough to disable the watchdog for my firmware.

It will throw an error if either the watchdog is enabled (Bit 6) in the sync manager control register 0x0804, or if 0x0420 is non-zero. So clearing bit 6 in 0x0804 might be enough to make it work.

I'll test that out tomorrow.

Cheers, m.

Apollo3zehn, 2020-03-01:
Okay, I`ll wait then for your feedback.

closed 2020-03-02 by mschwerin