XzuluX/EtherCAT.NET #43: SampleMaster Read PDO

Nick135, 2020-03-03:
I have a very simple Projekt with only one Slave (EP3174-0002 4 Channel Analog Input).
The EP is found and the Master seems to run.

The 'slaves[0].DynamicData.PdoSet[0].VariableSet' contains the different PDOs and they also get a Pointer Address to the RAM.
The 'slaves[0].DynamicData.PdoSet[0].VariableSet.Last()' is the Value.

In the 'README.md' is only how to set a single bit per unsafe.
How can I get the Value form the PDO (Pointer).
Is there a simpler way to cyclic get the input value?

PS: Is there a implementation to write slave settings with command or on 'startup'?

Apollo3zehn, 2020-03-03:
I see three options to get the value:

### Option 1
When you know the data type of the variable, you can access the value via:

```cs
var myVariableSpan = new Span<int>(variable0.DataPtr.ToPointer(), 1);
var myValue = myVariableSpan.First();
```

### Option 2
If you prefer to avoid the unsafe keyword in combination with ```Span<T>```, you can instead copy the raw bytes from unmanged memory to managed memory:

```cs
var rawData = new byte[4];
Marshal.Copy(variables.Last().DataPtr, rawData, 0, 4);
var value = BitConverter.ToInt32(rawData);
```

### Option 3
Or you can create an extension method for the variable which must be located in a static class:

```cs
public static T GetValue<T>(this SlaveVariable variable)
{
    unsafe
    {
        var span = new Span<T>(variable.DataPtr.ToPointer(), 1);
        return span[0];
    }
}
```

With this extension method, you can access the variable`s value via:

```cs
var myValue = variable.GetValue<int>();
```

I have not yet implemeted any easier method to access the data because I would first like to rewrite the whole project to use C# only (i.e. translate the underlying SOEM library to C#, if possible).

### Regarding slave settings on startup:

You can register a callback function to be called on startup (PreOP -> SafeOP) with:
```cs
var slaveIndex = 1; // the first slave has index 1 since 0 is the master itself

EcHL.RegisterCallback(master.Context, slaveIndex, slaveIndex =>
{
    var returnValue = EcUtilities.SdoWrite(this.Context, slaveIndex, ...);
    EcUtilities.CheckErrorCode(master.Context, returnValue);

    return 0;
});
```
The ```SdoWrite(...)``` method within the callback is used to send SDO commands to the specified slave to modify its configuration before the EtherCAT network starts.

Here is an example how the master itself registers the callbacks: https://github.com/Apollo3zehn/EtherCAT.NET/blob/master/src/EtherCAT.NET/EcMaster.cs#L142-L152

And here are the lines of code where the underyling SOEM master calls the registered callback functions: https://github.com/OpenEtherCATsociety/SOEM/blob/093311561c36810cb16957850b3a13ad9ddf4887/soem/ethercatconfig.c#L651-L655

Nick135, 2020-03-04:
I start the Project with Windows, but at the end it sould run on Linux (Raspberry if possible).

On Windows it is working fine.
On Linux (Ubuntu Mate 18.04.3 x64) I get only two packets from the master and then there is a Arithmetic error on 'master.UpdateIO(DateTime.UtcNow);'

Is there a way to run vs code with root property in the launch.json (this works with pySOEM, but with OmniSharp "sudo": true is unknown)

`osboxes@osboxes:~/_Projekt/Apollo3zehn/EtherCAT.NET/artifacts/bin/SampleMaster/netcoreapp3.0$ sudo dotnet '/home/osboxes/_Projekt/Apollo3zehn/EtherCAT.NET/artifacts/bin/SampleMaster/netcoreapp3.0/SampleMaster.dll'
info: EtherCAT Master[0]
      Found 1 slaves:
      EP3174-0002 (PDOs: 8 - CSA: 62808)
info: EtherCAT Master[0]
      IO map configured (1 slave, 16 bytes)
info: EtherCAT Master[0]
      DC system time diff. is <= 0 ns
info: EtherCAT Master[0]
      DC drift compensation started
dbug: EtherCAT Master[0]
      Value: inputValue=16406; Value=5,006866664632099
dbug: EtherCAT Master[0]
      Value: inputValue=16412; Value=5,008697775200659
dbug: EtherCAT Master[0]
      Value: inputValue=16458; Value=5,022736289559618
fail: EtherCAT Master[0]
      System.OverflowException: Arithmetic operation resulted in an overflow.
         at System.Linq.Enumerable.Average(IEnumerable`1 source)
         at EtherCAT.NET.EcMaster.UpdateIO(DateTime referenceDateTime) in /home/osboxes/_Projekt/Apollo3zehn/EtherCAT.NET/src/EtherCAT.NET/EcMaster.cs:line 378
         at SampleMaster.Program.<>c__DisplayClass0_2.<Main>b__3() in /home/osboxes/_Projekt/Apollo3zehn/EtherCAT.NET/sample/SampleMaster/Program.cs:line 120
fail: EtherCAT Master[0]
      System.OverflowException: Arithmetic operation resulted in an overflow.
         at System.Linq.Enumerable.Average(IEnumerable`1 source)
         at EtherCAT.NET.EcMaster.UpdateIO(DateTime referenceDateTime) in /home/osboxes/_Projekt/Apollo3zehn/EtherCAT.NET/src/EtherCAT.NET/EcMaster.cs:line 378
         at SampleMaster.Program.<>c__DisplayClass0_2.<Main>b__3() in /home/osboxes/_Projekt/Apollo3zehn/EtherCAT.NET/sample/SampleMaster/Program.cs:line 120
`

Nick135, 2020-03-04:
This is when the first error occurs.

`fail: EtherCAT Master[0]
      System.Exception: Arithmetic operation resulted in an overflow.: _dcRingBuffer:[-2711694567532672400,-2711694567531745500,-2711694567531652800,-2711694567531565700,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0]
       ---> System.OverflowException: Arithmetic operation resulted in an overflow.
         at System.Linq.Enumerable.Average(IEnumerable`1 source)
         at EtherCAT.NET.EcMaster.UpdateIO(DateTime referenceDateTime) in /home/osboxes/_Projekt/Apollo3zehn/EtherCAT.NET/src/EtherCAT.NET/EcMaster.cs:line 382
         --- End of inner exception stack trace ---
         at EtherCAT.NET.EcMaster.UpdateIO(DateTime referenceDateTime) in /home/osboxes/_Projekt/Apollo3zehn/EtherCAT.NET/src/EtherCAT.NET/EcMaster.cs:line 387
         at SampleMaster.Program.<>c__DisplayClass0_2.<Main>b__3() in /home/osboxes/_Projekt/Apollo3zehn/EtherCAT.NET/sample/SampleMaster/Program.cs:line 120
`

`                // the UpdateIo timer tries to fire at discrete times. The timer is allowed to be early or delayed by < (CycleTime - Offset) and the resulting DC time will be floored to nearest 10 ms. 

                this.UtcDateTime = _dcEpoch.AddTicks(Convert.ToInt64(_dcTime / _dateTime_To_Ns));  // for time loop control

                // dc drift compensation
                _dcRingBuffer[_dcRingBufferIndex] = Convert.ToInt64(referenceDateTime.Ticks - this.UtcDateTime.Ticks) * _dateTime_To_Ns;
                try
                {
                    this.DcRingBufferAverage = Convert.ToInt64(_dcRingBuffer.Average());
                }
                catch(Exception ex)
                {
                    throw new Exception($"{(ex.InnerException ?? ex).Message}: _dcRingBuffer:{(_dcRingBuffer !=null ? "[" + string.Join(",", _dcRingBuffer.Select(s=>s))+ "]" : "null")}", (ex.InnerException ?? ex));
                }`

Apollo3zehn, 2020-03-04:
I guess in your master code you still pass a UTC time? ```master.UpdateIO(DateTime.UtcNow);```. I can only imagine that the wrong date time is passed to the UpdateIO method. Another reason for an overflow can bei very large lags (e.g. with a Debugger attached). I will later have a look to this issue.

Apollo3zehn, 2020-03-04:
I think I already solved the issue within my PR for the next alpha version (https://github.com/Apollo3zehn/EtherCAT.NET/pull/36/commits/006d7650578f4ecd6835fafa31569f44465da512)

The problem is that on Linux SOEM is not using the OS clock to initialize the distributed clock time. This master then compares the OS time to a random number, which may result in very large numbers that overflow eventually. The same happened on Windows but the behavior was changed for the current version of SOEM (https://github.com/OpenEtherCATsociety/SOEM/pull/283)

I will later today publish a new alpha version. Hopefully this already solves your issues.

Regarding OmniSharp: I have not worked on Linux with EtherCAT for a while so I do not know how to run this with sudo, but maybe this helps? https://github.com/OmniSharp/omnisharp-vscode/issues/1811

Apollo3zehn, 2020-03-04:
I published 1.0.0-alpha.3.final and the master branch has been updated also. Please test if with this version the problem persists. Thank you :)

Nick135, 2020-03-04:
Update are working fine. Thanks.

To Debug with vs code, this works for me:
sudo code --user-data-dir=path

Apollo3zehn, 2020-03-06:
Thanks for your feedback. I'm closing this issue as it's been solved.

closed 2020-03-06 by Apollo3zehn