XzuluX/EtherCAT.NET #44: Compile on ARM (Raspberry Pi 4)

Nick135, 2020-03-06:
I install the compiler and change the init_solution.ps1 (remove -m32 at X86 and the 64Bit part)

Now I get this error:
`ubuntu@ubuntu:~/#Projekt/EtherCAT.NET-1.0.0-alpha.3.final$ pwsh init_solution.ps1
Updating Git submodule.
fatal: not a git repository (or any of the parent directories): .git
Creating native x86 project.

    Directory: /home/ubuntu/#Projekt/EtherCAT.NET-1.0.0-alpha.3.final/artifacts

Mode                 LastWriteTime         Length Name
----                 -------------         ------ ----
d----          03/06/2020    14:54                bin32
-- The C compiler identification is GNU 7.4.0
-- The CXX compiler identification is GNU 7.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at CMakeLists.txt:11 (add_subdirectory):
  The source directory

    /home/ubuntu/#Projekt/EtherCAT.NET-1.0.0-alpha.3.final/native/SOEM

  does not contain a CMakeLists.txt file.

-- Configuring incomplete, errors occurred!
See also "/home/ubuntu/#Projekt/EtherCAT.NET-1.0.0-alpha.3.final/artifacts/bin32/CMakeFiles/CMakeOutput.log".
`

Apollo3zehn, 2020-03-06:
Does the repository contain a .git repository? What does 'git status' return if run from within the repository?

Best would be to try a fresh repo:

```cs
git clone https://github.com/Apollo3zehn/EtherCAT.NET
cd EtherCAT.NET

// ... make your modifications to init_solution.ps1

.\init_solution.ps1
```

Nick135, 2020-03-06:
This work to compile (instead of the release zip folder)

Now I get this Error:
`ubuntu@ubuntu:~/#Projekt/20200306/EtherCAT.NET$ dotnet build ./src/EtherCAT.NET/EtherCAT.NET.csproj
Microsoft (R) Build Engine version 16.4.0+e901037fe for .NET Core
Copyright (C) Microsoft Corporation. All rights reserved.

  Restore completed in 141.45 ms for /home/ubuntu/#Projekt/20200306/EtherCAT.NET/src/SOEM.PInvoke/SOEM.PInvoke.csproj.
  Restore completed in 155.97 ms for /home/ubuntu/#Projekt/20200306/EtherCAT.NET/src/EtherCAT.NET/EtherCAT.NET.csproj.
/home/ubuntu/dotnet/sdk/3.1.102/Microsoft.Common.CurrentVersion.targets(4601,5): error MSB3030: Could not copy the file "/home/ubuntu/#Projekt/20200306/EtherCAT.NET/artifacts/bin64/SOEM_wrapper/libsoem_wrapper.so" because it was not found. [/home/ubuntu/#Projekt/20200306/EtherCAT.NET/src/SOEM.PInvoke/SOEM.PInvoke.csproj]

Build FAILED.

/home/ubuntu/dotnet/sdk/3.1.102/Microsoft.Common.CurrentVersion.targets(4601,5): error MSB3030: Could not copy the file "/home/ubuntu/#Projekt/20200306/EtherCAT.NET/artifacts/bin64/SOEM_wrapper/libsoem_wrapper.so" because it was not found. [/home/ubuntu/#Projekt/20200306/EtherCAT.NET/src/SOEM.PInvoke/SOEM.PInvoke.csproj]
    0 Warning(s)
    1 Error(s)

Time Elapsed 00:00:06.58
`

closed 2020-03-06 by Nick135