XzuluX/EtherCAT.NET #45: DynamicData PdoSet

Nick135, 2020-03-08:
How works the DynamicData?

I have a Slave (EP3174-0002) with 4 Channel.
Each channel has a Standard and a Compact PDO Setting.
In my case the lib use the Standard PDO. Why?

Is there a Parameter to detect the channel count or only by the "name" parameter?

[image]

[image]

Apollo3zehn, 2020-03-09:
As far as I know there is no parameter to detect the number of channels. EtherCAT itself simply exchanges process data and has no concept of channels.

The dynamic data are constructed using the provided ESI information. Here is a part of the ESI for your slave:

[image]

It shows the first two PDO that are also visible in your screenshot. The first PDO is assigned to sync manager 3, the second PDO is not assigned to any sync manager by default. 

Another important thing to note is that the both PDO's exclude each other. When you enable the first PDO, it is not possible to enable the second PDO (see the ```<Exclude>...<Exclude/>``` lines within the ESI).

I am not 100% sure but you can try to set the first PDO's sync manager to -1 and second PDO's sync manager to 3. Hopefully after this, everything is configured correctly to use only the second PDO:

```cs
rootSlaveInfo.Descendants().ToList().ForEach(current =>
{
      ExtensibilityHelper.CreateDynamicData(settings.EsiDirectoryPath, extensionFactory, current);
});

// ...

slaves[0].DynamicData.PdoSet[0].SyncManager = -1;
slaves[0].DynamicData.PdoSet[1].SyncManager = 3;

// ...

master.Configure(rootSlaveInfo);
```

joaosbastos, 2020-03-12:
Hi,
i'm trying to connect to Helmholz TB20-C Ethercat and managed to get the communication running. I had to activate a parameter on the gateway that said to report the connected modules so that the connection reached operational state, without that it didn't go through safe-operational.
My problem is that my PdoSet array is empty, this is a gateway with I/O modules i can see the gateway but not the modules. Do i have to edit the ESI in any way? Or can i access the modules in any "raw" mode? 
[TB20_EtherCAT_ESI_V1.04.002.zip](https://github.com/Apollo3zehn/EtherCAT.NET/files/4325969/TB20_EtherCAT_ESI_V1.04.002.zip)

Apollo3zehn, 2020-03-13:
Here is an excerpt of your ESI file:

[image]

It shows that your gateway does not have direct PDO definitions but uses modules instead. Unfortunately I have not yet implemented a way to work with these module definitions. 

You can find more information here (https://www.ethercat.org/en/downloads/downloads_715D4CF690F241FDAD437BD1C3E0E5E4.htm) and here (https://www.ethercat.org/download/documents/ETG5001_6220_V1i0i5_S_R_IO-LinkMaster.pdf).

My focus within the next months is to create a purely managed EtherCAT master without depending on SOEM. Until then I will not implement new features, sorry!

The only workaround is to go through your device description and create a plugin like I did here for the Profibus Gateway which also uses modules: https://github.com/Apollo3zehn/EtherCAT.NET/tree/master/src/EtherCAT.NET/Extension/EL6731-0010 (ignore the WebClient folder)

When you have created your plugin, you can apply it like this:

```cs
// ...
var rootSlaveInfo = EcUtilities.ScanDevices(settings.InterfaceName);
var slaves = rootSlaveInfo.Descendants().ToList();

// slave[0] is the master
slaves[1].SlaveExtensionSet = new List<SlaveExtensionSettingsBase>() { new EL6731_0010Settings(slaves[1]) };

rootSlaveInfo.Descendants().ToList().ForEach(current =>
{
    ExtensibilityHelper.CreateDynamicData(settings.EsiDirectoryPath, extensionFactory, current);
});

// ...
```

In case you have a fixed set of modues the easiest way to create a plugin would be to use TwinCAT's startup tab to find out the SDO commands for your current configuration:

[image]

Your plugin would then need to return the correct SDO command*:
https://github.com/Apollo3zehn/EtherCAT.NET/blob/master/src/EtherCAT.NET/Extension/EL6731-0010/EL6731_0010Extension.cs#L100-L103

* Maybe you can skip this part as your gateway has hardware modules attached and therefore it does not need to be configured manually. But it may happen that you gateway requires the master to confirm the current SDO configuration by writing it to the slave. In the first case you can simply return an empty SDO command list, in the second case you need to find out the SDO command as written above.

And then, you need to go through you ESI file, find out the module's PDO addresses for your current configuration and add the PDO's and variables to your slave: https://github.com/Apollo3zehn/EtherCAT.NET/blob/master/src/EtherCAT.NET/Extension/EL6731-0010/EL6731_0010Settings.cs#L130-L139

If you attach the plugin to the slave as shown above, you should get your working PDO's and variables.

closed 2020-03-17 by Nick135