XzuluX/EtherCAT.NET #46: Unable to load DLL 'soem_wrapper' or one of its dependencies

Nick135, 2020-03-11:
The Application is running on my build PC.

On the device pc I get the unable load exception.
Device is Windows 10 Pro  x64:
 - WinPcap 4.1.3
 - DotNet Core Desktop Runtime 3.1

Apollo3zehn, 2020-03-11:
Maybe you also need the C++ Runtime:

https://support.microsoft.com/en-us/help/2977003/the-latest-supported-visual-c-downloads

Nick135, 2020-03-12:
That works. Thanks

The Connection is working good, but it need ca. 18 seconds on startup. Is this normal?

Apollo3zehn, 2020-03-12:
This probably comes from these lines of code:
https://github.com/Apollo3zehn/EtherCAT.NET/blob/master/native/SOEM_wrapper/soem_wrapper.c#L425-L428

When the system starts, the distributed clocks of all slaves are synchronized by sending a mass amount of synchronization frames to the network.

If you don't need high DC accuracy, you can reduce the number of frames (default is 15000) to a number you prefer, e.g.:

```
// ...

/* create EtherCAT master settings (with 10 Hz cycle frequency) */
var cycleFrequency = 10U;
var settings = new EcSettings(cycleFrequency, esiDirectoryPath, interfaceName)
{
    FrameCount = 1000
}

// ...
```

The actual reason for this being slow is probably your network card, which throttles the frames. See #7 for more information.

Nick135, 2020-03-12:
> Computer\HKEY_LOCAL_MACHINE\SYSTEM\ControlSet001\Control\Class\{4d36e972-e325-11ce-bfc1-08002be10318}\0001
> *InterruptModeration = 0
No change

Reduce FrameCount works

Nick135, 2020-03-17:
As I know DC is to synchronize the slaves to the same clock. For the DC master Beckhoff use the first slave with DC functionality, because the ethernet controler of the ipc is not exact enough.

In my case I don´t need DC. I als have only one slave, so I think DC make no sense.
Can I dissable DC?
DC shift compensation starts from time to time. Make this senso for only one slave?

Apollo3zehn, 2020-03-18:
If your slave doesn't use oversampling, DC is probably not required for your measurement. 

I do not prefer to disable DC completely because many slaves need it and activating it by default makes them simply work. 

If you are compiling the application by yourself, you could remove these lines of code in the master:
https://github.com/Apollo3zehn/EtherCAT.NET/blob/master/src/EtherCAT.NET/EcMaster.cs#L374-L403

Nick135, 2020-03-18:
Not disable completely, disable optional.

I think oversampling (with hard realtime) and C# is not a good idea. Do you test with 1ms bus speed and oversampling (e.g. 50kSps)?

Apollo3zehn, 2020-05-28:
Typically I am running EtherCAT with 10 ms bus speed which works fine with the process set to realtime priority (on Windows). Also oversampling worked fine in my tests. Now that Preempt-RT became more integrated into the Linux kernel, I would like to give it a try and test 1 ms bus speed or more.

I´ll close this issue because the actual problem (missing DLL) is solved.

closed 2020-05-28 by Apollo3zehn