XzuluX/EtherCAT.NET #47: error:06070013 Data type does not match, length of service parameter too low

GithubTwister, 2020-04-09:
Hello,

i have Problem with EtherCAT Slave  "PD4-EB59CD-E-65-1" (Nanotec Motor).
The ESI-File should be correct because TwinCAT3 works with it fine.

If i starte SampleMaster i get the following Messages:
===========================================================
info: EtherCAT Master[0]
      Found 4 slaves:
      EK1100 (PDOs: 0 - CSA: 36726)
      EL2828 (PDOs: 8 - CSA: 9020)
      EL7047 (PDOs: 19 - CSA: 48200)
      PD4-EB59CD-E-65-1 (PDOs: 8 - CSA: 32014)
fail: EtherCAT Master[0]
      SdoWrite failed (0x0000): The working counter is 0 (SOEM).

EtherCAT message:

Time:1586370400.977 SDO slave:4 index:1a00.01 error:06070013 Data type does not match, length of service parameter too low

Time:1586370400.979 SDO slave:4 index:1a00.02 error:06070013 Data type does not match, length of service parameter too low

Time:1586370400.981 SDO slave:4 index:1a00.03 error:06070013 Data type does not match, length of service parameter too low

Time:1586370400.982 SDO slave:4 index:1a00.04 error:06070013 Data type does not match, length of service parameter too low

Time:1586370400.985 SDO slave:4 index:1a00.05 error:06070013 Data type does not match, length of service parameter too low

Time:1586370400.987 SDO slave:4 index:1a00.06 error:06070013 Data type does not match, length of service parameter too low
===========================================================
The error occurs in the following line:
In file "Program.cs"
 master.Configure(rootSlaveInfo);

  In File "EcMaster.cs"
  this.ConfigureIoMap(slaveInfoSet);  
      
     _actualIoMapSize = EcHL.ConfigureIoMap(this.Context, _ioMapPtr, slaveRxPdoOffsetSet, slaveTxPdoOffsetSet, out _expectedWorkingCounter);

What can i do to find the reason of the error and solve the problem?

Regards

Apollo3zehn, 2020-04-09:
Could you please provide the ESI file of your slave? I could not find it on their homepage.

GithubTwister, 2020-04-11:
[PD4-EB59CD-E-65-1.zip](https://github.com/Apollo3zehn/EtherCAT.NET/files/4465503/PD4-EB59CD-E-65-1.zip)

Apollo3zehn, 2020-04-12:
Maybe the root cause is here:

https://github.com/Apollo3zehn/EtherCAT.NET/blob/5ea922a0101636ed17a41563c958a7db5b34a032/src/EtherCAT.NET/Infrastructure/SlaveInfo.cs#L156-L160

And I found this statement in the ETG forum: ```PDO Mapping objects are always UINT32```.

To be honest, I am not sure why I am adding ```UINT16``` objects to the SDO write request (its long ago it wrote that code part and I would like to rewrite anything this year). So maybe it will help if you change this code to:

```cs
foreach (SlaveVariable slaveVariable in slavePdo.VariableSet)
{
    var lowBytes = (ushort)(slaveVariable.SubIndex + (Convert.ToUInt16(slaveVariable.BitLength) << 8));
    var highBytes = slaveVariable.Index;
    dataset.Add((uint)(lowBytes + (highBytes << 16)));
}
```

Maybe this code will only work with your motor slave, so please test it first with that slave only. And maybe the "lowBytes" and "highBytes" variables should be interchanged.

I could also be helpful If you provide Wireshark logs from the EtherCAT network startup with TwinCAT and EtherCAT.NET (to compare both).

GithubTwister, 2020-04-12:
Ok, i made the changes
1)With only one Slave "PD4-EB59CD-E-65-1"
=====================================================
info: EtherCAT Master[0]
      Found 1 slaves:
      PD4-EB59CD-E-65-1 (PDOs: 8 - CSA: 32014)
info: EtherCAT Master[0]
      IO map configured (1 slave, 18 bytes)
info: EtherCAT Master[0]
      DC system time diff. is <= 0 ns
fail: EtherCAT Master[0]
      CheckSafeOpState failed (0x0501): One or more terminals did not reach SafeOp-state.
=====================================================

2) With all Slaves
=====================================================
info: EtherCAT Master[0]
      Found 4 slaves:
      EK1100 (PDOs: 0 - CSA: 36726)
      EL2828 (PDOs: 8 - CSA: 9020)
      EL7047 (PDOs: 19 - CSA: 48200)
      PD4-EB59CD-E-65-1 (PDOs: 8 - CSA: 32014)
info: EtherCAT Master[0]
      IO map configured (4 slaves, 23 bytes)
info: EtherCAT Master[0]
      DC system time diff. is <= 15 ns
fail: EtherCAT Master[0]
      CheckSafeOpState failed (0x0501): One or more terminals did not reach SafeOp-state.

EtherCAT message:

Time:1586720780.480 SDO slave:3 index:0216.00 error:06010001 Attempt to read to a write only object

Time:1586720780.481 SDO slave:3 index:0416.00 error:06010001 Attempt to read to a write only object

Time:1586720780.484 SDO slave:3 index:031a.00 error:06010001 Attempt to read to a write only object
=====================================================

3) With the other Slaves
EK1100 (PDOs: 0 - CSA: 36726)
EL2828 (PDOs: 8 - CSA: 9020)
EL7047 (PDOs: 19 - CSA: 48200)
=====================================================
info: EtherCAT Master[0]
      Found 3 slaves:
      EK1100 (PDOs: 0 - CSA: 36726)
      EL2828 (PDOs: 8 - CSA: 9020)
      EL7047 (PDOs: 19 - CSA: 48200)
info: EtherCAT Master[0]
      IO map configured (3 slaves, 5 bytes)
info: EtherCAT Master[0]
      DC system time diff. is <= 0 ns
info: EtherCAT Master[0]
      DC drift compensation started
info: EtherCAT Master[0]
      DC drift compensation finished
....
=====================================================

I do not have Wireshark.
Do I need it to solve the problem?

Regards

Apollo3zehn, 2020-04-13:
Thanks for that analysis. I think we should focus only on the motor slave first.

Did you also try to exchange the high and low bytes, i.e:
```cs
foreach (SlaveVariable slaveVariable in slavePdo.VariableSet)
{
    var highBytes = (ushort)(slaveVariable.SubIndex + (Convert.ToUInt16(slaveVariable.BitLength) << 8));
    var lowBytes = slaveVariable.Index;
    dataset.Add((uint)(lowBytes + (highBytes << 16)));
}
```
What is the output then?

The wireshark logs would be very useful. It is free and works both on Windows and Linux. It captures your network/EtherCAT traffic. To capture the traffic with TwinCAT you need to put the adapter into promiscuous mode first: https://infosys.beckhoff.com/index.php?content=../content/1031/tf6281_tc3_ethernetipscanner/4350142731.html&id=

And please be aware that if TwinCAT is running you cannot run EtherCAT.NET on the same computer because TwinCAT blocks the network card access. Put TwinCAT to configuration mode first (only if both applications run on the same computer).

GithubTwister, 2020-04-13:
Yes, i tried it before
I saw that IO-map was 69 Bytes, so i know that your first Suggestion was Right.

 with follwoing code
```
foreach (SlaveVariable slaveVariable in slavePdo.VariableSet)
                        {
                            var lowBytes = (ushort)(slaveVariable.SubIndex + (Convert.ToUInt16(slaveVariable.BitLength) << 8));
                            var highBytes = slaveVariable.Index;
                            dataset.Add((uint)(highBytes + (lowBytes << 16)));
                        }

```
i got this message

> info: EtherCAT Master[0]
>       Found 1 slaves:
>       PD4-EB59CD-E-65-1 (PDOs: 8 - CSA: 32014)
> info: EtherCAT Master[0]
>       IO map configured (1 slave, 69 bytes)
> info: EtherCAT Master[0]
>       DC system time diff. is <= 0 ns
> fail: EtherCAT Master[0]
>       CheckSafeOpState failed (0x0501): One or more terminals did not reach SafeOp-state.

GithubTwister, 2020-04-13:
Now i made a Wireshark file

[Wireshark_Nanotec.zip](https://github.com/Apollo3zehn/EtherCAT.NET/files/4468733/Wireshark_Nanotec.zip)

Apollo3zehn, 2020-04-13:
Thanks for the log. According to it, the byte order written to the SDO object is incorrect:

[image]

It should be ```0x10004160``` but is ```0x60411000```.

Could you create another log with the following code and also post the console output?

```cs
foreach (SlaveVariable slaveVariable in slavePdo.VariableSet)
{
    var bytes = BitConverter.GetBytes(slaveVariable.Index);
    Array.Reverse(bytes);
    var index = BitConverter.ToUInt16(bytes, 0);

    var lowBytes = index;
    var highBytes = (ushort)(slaveVariable.SubIndex + (Convert.ToUInt16(slaveVariable.BitLength) << 8));
                        
    dataset.Add((uint)(lowBytes + (highBytes << 16)));
}
```

Thank you!

GithubTwister, 2020-04-13:
> info: EtherCAT Master[0]
>       Found 1 slaves:
>       PD4-EB59CD-E-65-1 (PDOs: 8 - CSA: 32014)
> info: EtherCAT Master[0]
>       IO map configured (1 slave, 79 bytes)
> info: EtherCAT Master[0]
>       DC system time diff. is <= 0 ns
> fail: EtherCAT Master[0]
>       CheckSafeOpState failed (0x0501): One or more terminals did not reach SafeOp-state.

[Wireshark_Nanotec2.zip](https://github.com/Apollo3zehn/EtherCAT.NET/files/4469071/Wireshark_Nanotec2.zip)

> 
> "SampleMaster.exe" (CoreCLR: clrhost): "C:\Program Files\dotnet\shared\Microsoft.NETCore.App\3.1.3\System.Security.Cryptography.Algorithms.dll" geladen. Das Laden von Symbolen wurde übersprungen. Das Modul ist optimiert, und die Debugoption "Nur eigenen Code" ist aktiviert.
> Statusword 43/0
> Position actual value 45/0
> Modes of operation display 49/0
> Controlword 0/0
> Target Position 2/0
> Motor drive submode select 6/0
> Modes of operation 10/0
> Ausnahme ausgelöst: "System.Exception" in EtherCAT.NET.dll
> Ausnahme ausgelöst: "System.Exception" in SampleMaster.dll
> Eine Ausnahme vom Typ "System.Exception" ist in SampleMaster.dll aufgetreten, doch wurde diese im Benutzercode nicht verarbeitet.
> CheckSafeOpState failed (0x0501): One or more terminals did not reach SafeOp-state.
> 
> Das Programm "[4380] SampleMaster.exe" wurde mit Code -1 (0xffffffff) beendet.
>

GithubTwister, 2020-04-13:
I tried this:
And it works fine.
Thank you very much...

```

foreach (SlaveVariable slaveVariable in slavePdo.VariableSet)
                        {
                            var lowBytes = slaveVariable.Index;
                            var highBytes = (ushort)(slaveVariable.SubIndex + (Convert.ToUInt16(slaveVariable.BitLength) << 8));

                            var bytes = BitConverter.GetBytes(highBytes);
                            Array.Reverse(bytes);
                            highBytes = BitConverter.ToUInt16(bytes, 0);

                            dataset.Add((uint)(highBytes + (lowBytes << 16)));
                        }
```

Apollo3zehn, 2020-04-13:
And again I confused the byte order :) I have nothing to test right now, good you found the final solution by yourself. 

Do all slaves work now? Or is there still an issue with slave 3 (EL7047). I am wondering why the error message is like ```... index:0216.00 error:06010001 ...```. The index should be reversed, i.e. ```... index:1602.00 error ...```. But the ESI file is fine, so maybe its just the slave that provides the wrong index in response to the illegal read attempt.

GithubTwister, 2020-04-13:
Yes, all slaves are working correct now, without error Messages.

Now I can start to programm the communication to the motor-slave.
Is there a easy and clear way to access to the PDOs?
Like "slaves[3].PDO[0].SetValue(10);" ?
Or do i have to use the tricky way like in sample?
                                
```
                                var myVariableSpan = new Span<int>(variables.First().DataPtr.ToPointer(), 1);
                                myVariableSpan[0] = random.Next(0, 255);
```

Apollo3zehn, 2020-04-13:
Since I am planning to rewrite the whole code (to make things much easier, such as PDO access), I am not adding new features right now. But you can create a small Extension method as described here: https://github.com/Apollo3zehn/EtherCAT.NET/issues/43#issuecomment-594221641

GithubTwister, 2020-04-13:
Ok, thank you, i will find a way. :-)

Apollo3zehn, 2020-05-12:
I´ll close this for now, please feel free to reopen if anything is unclear.

commit ad44c4f2 (2020-05-28): Fix #47 and #49. (#50)

* Improve DC init error message.

* Properly parse HexDec strings.

* Improve code style.

* Fix missing subindex padding bug.

* Prepare release.
closed 2020-05-12 by Apollo3zehn