XzuluX/EtherCAT.NET #48: CreateDynamicData failing to load ESI

joaosbastos, 2020-05-19:
On this ESI i'm getting this error:
`startIndex cannot be larger than length of string.
Parameter name: startIndex :    at System.String.Substring(Int32 startIndex, Int32 length)
   at EtherCAT.NET.Extensibility.ExtensibilityHelper.<>c__DisplayClass0_1.<CreateDynamicData>b__1(PdoTypeEntry x) in C:\projects\ethercat-net\src\EtherCAT.NET\Extensibility\ExtensibilityHelper.cs:line 87
   at System.Linq.Enumerable.WhereSelectArrayIterator`2.MoveNext()
   at System.Collections.Generic.List`1..ctor(IEnumerable`1 collection)
   at System.Linq.Enumerable.ToList[TSource](IEnumerable`1 source)
   at EtherCAT.NET.Extensibility.ExtensibilityHelper.CreateDynamicData(String esiDirectoryPath, IExtensionFactory extensionFactory, SlaveInfo slaveInfo) in C:\projects\ethercat-net\src\EtherCAT.NET\Extensibility\ExtensibilityHelper.cs:line 93`

[Festo-CMMT-ST-CiA402-20200113.txt](https://github.com/Apollo3zehn/EtherCAT.NET/files/4649447/Festo-CMMT-ST-CiA402-20200113.txt)

Apollo3zehn, 2020-05-19:
Could you please try to replace `<Index>0</Index>` by `<Index>#x0</Index>` in the ESI file? This converts the decimal zero into a hexadecimal zero. This is a bug in EtherCAT.NET, but as I said in the other issues, I want to rewrite the whole code (and then properly parse decimal and hexadecimal numbers). Until then, I only collect potential errors but I am not fixing them right now. I hope the workaround lets you continue.

Please be aware that EtherCAT.NET uses an ESI cache. So please first delete the content of `%localappdata%\EtherCAT.NET\cache`.

joaosbastos, 2020-05-19:
After correcting de index from decimal to hex, i'm getting another error:
`ConfigureDc failed (0x0302): The static drift compensation for distributed clocks failed.

EtherCAT message:

Time:1589896047.595 SDO slave:2 index:4016.00 error:06020000 The object does not exist in the object directory

Time:1589896047.601 SDO slave:2 index:401a.00 error:06020000 The object does not exist in the object directory

Time:1589896047.618 SDO slave:2 index:4016.00 error:06020000 The object does not exist in the object directory

Time:1589896047.635 SDO slave:2 index:401a.00 error:06020000 The object does not exist in the object directory

 :    at EtherCAT.NET.EcUtilities.CheckErrorCode(IntPtr context, Int32 errorCode, String callerMemberName) in C:\projects\ethercat-net\src\EtherCAT.NET\EcUtilities.cs:line 146
   at EtherCAT.NET.EcMaster.ConfigureDc() in C:\projects\ethercat-net\src\EtherCAT.NET\EcMaster.cs:line 228
   at EtherCAT.NET.EcMaster.Configure(SlaveInfo rootSlaveInfo) in C:\projects\ethercat-net\src\EtherCAT.NET\EcMaster.cs:line 292`

closed 2020-05-25 by joaosbastos

Apollo3zehn, 2020-05-26:
Sorry I did not see your message, but I have seen your new issue at the EtherCATSociety account, where fits better I think.