XzuluX/EtherCAT.NET #49: Index 0x1600 fails but 0x1601 accepts

joaosbastos, 2020-05-25:
My slave Input SM starts at 0x1600, and on my slave software it reads 0x0016 if i don't configure variables, if i add a variable on index 0x6040 to this SM i read 0x4016 on the SM index on the slave software. If i configure the SM index to 0x1601 on the slave software it reads 0x1601 correctly.
It seams as if the 00 from 0x1600 are being removed and the byte shifted right.

How can i solve this?

[Festo-CMMT-ST-CiA402-20200113.txt](https://github.com/Apollo3zehn/EtherCAT.NET/files/4678017/Festo-CMMT-ST-CiA402-20200113.txt)

Apollo3zehn, 2020-05-26:
For me it more seems to be a big-endian/little-endian confusion (instead of 0x1600 being shifted right).

There are the following PDOs defined in the ESI:
- `#x1600` (SM 2) - Outputs - 9 entries: Controlword, Modes of operation, Target position, ... 
- `#x1A00` (SM 3) - Inputs - 6 entries: Statusword, Modes of operation display, Position actual value, ...
- `#x1AF0` (not used by default) - Diag History - 1 entry: New Message Available
- `#x1AF1` (not used by default) - Timestamp - 1 entry: Timestamp

When you assign 0x1601 to the sync manager, I don't know what is happening, because it is not defined in the ESI. That may explain the differences in behavior between 0x1600 and 0x1601.

Did you try to write 0x0016 with the slave software? 

If you are using EtherCAT.NET and you get the error shown in #48, one thing you could try is to overwrite the written ESI content. Either you modify your ESI file and swap the bytes (`#x1600` to `#x0016` and delet the cache again), or you are modifing your program (from the sample):

```cs
/* create root slave info by scanning available slaves */
var rootSlaveInfo = EcUtilities.ScanDevices(settings.InterfaceName);

rootSlaveInfo.Descendants().ToList().ForEach(current =>
{
	ExtensibilityHelper.CreateDynamicData(settings.EsiDirectoryPath, extensionFactory, current);
});

/* modify the read data */
var slave = rootSlaveInfo.Descendants().First();
slave.DynamicData.PdoSet[0].Index = 0x0016 // or 0x1600
slave.DynamicData.PdoSet[1].Index = 0x001A // or 0x1A00
```
I cannot test this code right now, but maybe it helps.

joaosbastos, 2020-05-26:
I inverted the byte order on the ESI as attached and got the error:

`SdoWrite failed (0x0000): The working counter is 0 (SOEM).

EtherCAT message:

Time:1590485392.769 SDO slave:1 index:001a.00 error:06020000 The object does not exist in the object directory`

[Festo-CMMT-ST-CiA402-20200113.txt](https://github.com/Apollo3zehn/EtherCAT.NET/files/4681255/Festo-CMMT-ST-CiA402-20200113.txt)

Without the index byte order inverted:

`CheckSafeOpState failed (0x0501): One or more terminals did not reach SafeOp-state.

EtherCAT message:

Time:1590485972.998 SDO slave:1 index:4016.00 error:06020000 The object does not exist in the object directory

Time:1590485973.004 SDO slave:1 index:401a.00 error:06020000 The object does not exist in the object directory

Time:1590485973.021 SDO slave:1 index:4016.00 error:06020000 The object does not exist in the object directory

Time:1590485973.038 SDO slave:1 index:401a.00 error:06020000 The object does not exist in the object directory`

[Festo-CMMT-ST-CiA402-20200113_2.txt](https://github.com/Apollo3zehn/EtherCAT.NET/files/4681284/Festo-CMMT-ST-CiA402-20200113_2.txt)

Apollo3zehn, 2020-05-26:
So to recap everything:

In #48 and the unmodified ESI file you attached there, you get:
```
Time:1590155300.644 SDO slave:2 index:0016.00 error:06020000 The object does not exist in the object directory

Time:1590155300.651 SDO slave:2 index:001a.00 error:06020000 The object does not exist in the object directory
```

With inverted byte order, your get the same error (?):

```
Time:1590485392.769 SDO slave:1 index:001a.00 error:06020000 The object does not exist in the object directory`
```

And `Without the index byte order inverted`, i.e. now agian with the original ESI file, you get different errors than in #48 (?):

```
Time:1590485972.998 SDO slave:1 index:4016.00 error:06020000 The object does not exist in the object directory

Time:1590485973.004 SDO slave:1 index:401a.00 error:06020000 The object does not exist in the object directory
```

What is the difference between the first setup and the last? The ESI files are the same but the errors are different?

The best would be if you provide Wireshark captures, one with bytes swapped and one without. I guess then we are able to narrow down the issue.

Maybe the error is the EEPROM content. SOEM reads out the EEPROM and maybe there the PDO indices are wrong. But first a Wireshark capture would help.

joaosbastos, 2020-05-26:
In the first setup i cleared the dynamic data and wrote manually:

`ETHERCATSlaveInfo.Descendants().ToList()[0].DynamicData.PdoSet.Clear();

                //PDO Config OUTPUTS
                SlavePdo CMMTPdo_OUT = new SlavePdo(ETHERCATSlaveInfo, "Outputs", 0x1600, 10, true, true, 2); //SyncManager: 2-Saidas; 3-Entradas;
                List<SlaveVariable> ecCMMTVariableSet_OUT = new List<SlaveVariable>();
                ecCMMTVariableSet_OUT.Add(new SlaveVariable(CMMTPdo_OUT, "ControlWord", 0x6040, 0, DataDirection.Output, OneDasDataType.UINT16));
                ecCMMTVariableSet_OUT.Add(new SlaveVariable(CMMTPdo_OUT, "Modes of operation", 0x6060, 0, DataDirection.Output, OneDasDataType.INT8));
                ecCMMTVariableSet_OUT.Add(new SlaveVariable(CMMTPdo_OUT, "Target position", 0x607A, 0, DataDirection.Output, OneDasDataType.INT32));
                ecCMMTVariableSet_OUT.Add(new SlaveVariable(CMMTPdo_OUT, "Profile velocity", 0x6081, 0, DataDirection.Output, OneDasDataType.UINT32));
                ecCMMTVariableSet_OUT.Add(new SlaveVariable(CMMTPdo_OUT, "Target velocity", 0x60FF, 0, DataDirection.Output, OneDasDataType.INT32));
                ecCMMTVariableSet_OUT.Add(new SlaveVariable(CMMTPdo_OUT, "Target torque", 0x6071, 0, DataDirection.Output, OneDasDataType.INT16));
                ecCMMTVariableSet_OUT.Add(new SlaveVariable(CMMTPdo_OUT, "Velocity offset", 0x60B1, 0, DataDirection.Output, OneDasDataType.INT32));
                ecCMMTVariableSet_OUT.Add(new SlaveVariable(CMMTPdo_OUT, "Torque offset", 0x60B2, 0, DataDirection.Output, OneDasDataType.INT16));
                ecCMMTVariableSet_OUT.Add(new SlaveVariable(CMMTPdo_OUT, "", 0x0, 0, DataDirection.Output, OneDasDataType.BOOLEAN));
                CMMTPdo_OUT.SetVariableSet(ecCMMTVariableSet_OUT);
                ETHERCATSlaveInfo.Descendants().ToList()[0].DynamicData.PdoSet.Add(CMMTPdo_OUT);

                //PDO Config INPUTS
                SlavePdo CMMTPdo_IN = new SlavePdo(ETHERCATSlaveInfo, "Inputs", 0x1A00, 10, true, true, 3); //SyncManager: 2-Saidas; 3-Entradas;
                List<SlaveVariable> ecCMMTVariableSet_IN = new List<SlaveVariable>();
                ecCMMTVariableSet_IN.Add(new SlaveVariable(CMMTPdo_IN, "Statusword", 0x6041, 0, DataDirection.Input, OneDasDataType.UINT16));
                ecCMMTVariableSet_IN.Add(new SlaveVariable(CMMTPdo_IN, "Modes of operation display", 0x6061, 0, DataDirection.Input, OneDasDataType.INT8));
                ecCMMTVariableSet_IN.Add(new SlaveVariable(CMMTPdo_IN, "Position actual value", 0x6064, 0, DataDirection.Input, OneDasDataType.INT32));
                ecCMMTVariableSet_IN.Add(new SlaveVariable(CMMTPdo_IN, "Velocity actual value", 0x606C, 0, DataDirection.Input, OneDasDataType.INT32));
                ecCMMTVariableSet_IN.Add(new SlaveVariable(CMMTPdo_IN, "Torque actual value", 0x6077, 0, DataDirection.Input, OneDasDataType.INT16));
                ecCMMTVariableSet_IN.Add(new SlaveVariable(CMMTPdo_IN, "", 0x0, 0, DataDirection.Input, OneDasDataType.BOOLEAN));
                CMMTPdo_IN.SetVariableSet(ecCMMTVariableSet_IN);
                ETHERCATSlaveInfo.Descendants().ToList()[0].DynamicData.PdoSet.Add(CMMTPdo_IN);`

On the 2 other setups it came from the ESI.

joaosbastos, 2020-05-26:
I have made another test, i got a PLC from OMRON and syncronized all OK with the slave, and i got this output from the slave software:

[image]

Then i used the ESI without modifications with Ethercat.Net and got this output:

[image]

joaosbastos, 2020-05-26:
Wireshark capture with SM index unmodified:

[Wireshark_Unmodified.zip](https://github.com/Apollo3zehn/EtherCAT.NET/files/4681782/Wireshark_Unmodified.zip)

Wireshark capture with SM index 0x0016 and 0x001A:

[Wireshark_Modified.zip](https://github.com/Apollo3zehn/EtherCAT.NET/files/4681789/Wireshark_Modified.zip)

Apollo3zehn, 2020-05-26:
I think I found the issue. In your first Wireshark log you can see that EtherCAT.NET writes the (nearly) correct PDO addresses into registers 0x1C12 and 0x1C13, e.g. `010016` to map PDO 0x1600 to the sync manager at address 0x1C12. The only problem is that the number of PDOs (here: 0x01) should be an UINT16 instrad of UINT8, so the data written to 0x1600 must be `01000016`.

[image]

Are you able to compile EtherCAT.NET by yourself? If yes, please see this answer:
https://github.com/Apollo3zehn/EtherCAT.NET/issues/47#issuecomment-612643087

Please try to replace the code section mentiond in that comment with the following, finally working, code:

https://github.com/Apollo3zehn/EtherCAT.NET/issues/47#issuecomment-612844516 

If this solutions works (or you are not able to compile it yourself to test it first), I will release a small update to EtherCAT.NET to fix this because then it happened now two times in a row.

Apollo3zehn, 2020-05-26:
Ah sorry, this solution is related to the PDO mapping, not to the SM assignment. But the error is similar. I will look deeper into this.

Apollo3zehn, 2020-05-26:
Could you please try to change the `(byte)` to `(ushort)` in this line of code? https://github.com/Apollo3zehn/EtherCAT.NET/blob/5ea922a0101636ed17a41563c958a7db5b34a032/src/EtherCAT.NET/Infrastructure/SlaveInfo.cs#L184

This may cause other slaves to break. I need to find out what the rules are for the first subindex data type (UINT8 vs UINT16).

joaosbastos, 2020-05-26:
I changed to ushort and on the debug console i got the variable names, that i didn't. Now i got this error:
`CheckSafeOpState failed (0x0501): One or more terminals did not reach SafeOp-state. :    at EtherCAT.NET.EcUtilities.CheckErrorCode(IntPtr context, Int32 errorCode, String callerMemberName) in C:\Users\jbast\Downloads\EtherCAT.NET\src\EtherCAT.NET\EcUtilities.cs:line 148
   at EtherCAT.NET.EcMaster.Configure(SlaveInfo rootSlaveInfo) in C:\Users\jbast\Downloads\EtherCAT.NET\src\EtherCAT.NET\EcMaster.cs:line 298`

The wireshark capture:

[Wireshark_ushort.zip](https://github.com/Apollo3zehn/EtherCAT.NET/files/4683941/Wireshark_ushort.zip)

Apollo3zehn, 2020-05-26:
In the ETG Specs (ETG 1000.6 - Application Layer Protocol Specification, p. 80) - which I am not allowed to redistribute - there is written that the type of the first index is UNSIGNED8. And the type of the following PDO records is UNSIGNED16, i.e. the EtherCAT.NET code is correct. But it is not the first time I had this issue. I will tomorrow ask in the ETG forum what the reason for this inconsistency could be.

I will have a look to the remaining issue also tomorrow. Maybe now its related to the PDO mapping (see my previous comments).

joaosbastos, 2020-05-27:
Wireshark capture with (byte) to (ushort) and #47 modification:

[Wireshark_#47.zip](https://github.com/Apollo3zehn/EtherCAT.NET/files/4687264/Wireshark_.47.zip)

Apollo3zehn, 2020-05-27:
This looks much better. But I do not yet see why the slave is still not error-free.

joaosbastos, 2020-05-27:
The size and index of the SM appear now correct on the slave software, but the variables index is not correct. 
With the OMRON master i get this:

[image]

And now with the corrections:

[image]

On the outputs SM the first variable at index 0x6040 on the OMRON master reads 0x6040 0010‬,
but on Ethercat.NET it reads 0x8604000 .

Apollo3zehn, 2020-05-27:
That is a very good comparison. If translated to hex, the number of the first PDO in the first image is:
```hex(1614807056) = 0x60400010```

The number of the PDO in the second image is:
```hex(140525568) = 0x08604000```

And the transmitted data from the wireshark log:
```09 10 00 40 60 08 00 60 60 20 00 7a 60 ...```

So the slave extracts the byte sequence: 09 10 **00 40 60 08** 00 60 60 20 00 7a 60 ... instead of 09 **10 00 40 60** 08 00 60 60 20 00 7a 60 ....

According to the specification the string is correct (the data type of the first index is byte) but obviously the slave wants an extra byte again. In the code this means you shoud replace the `byte` by `ushort`:

https://github.com/Apollo3zehn/EtherCAT.NET/blob/5ea922a0101636ed17a41563c958a7db5b34a032/src/EtherCAT.NET/Infrastructure/SlaveInfo.cs#L154

joaosbastos, 2020-05-27:
It switched to RUN, thank you.
Do you want me to make any extra tests to this slave?
Just for reference this is FESTO CMMT-ST-C8-1C-EC-S0 firmware 17.0.7.78

Apollo3zehn, 2020-05-27:
Thanks for your feedback. I just asked in the ETG forum for an explanation of this behavior. I hope to get an answer soon so I can update this master. I think more tests are not required right now. I will later post a summary of the changes here that I would like you to confirm.

joaosbastos, 2020-05-27:
Is there an instruction to get the slave off operation (RUN)?

joaosbastos, 2020-05-27:
The manual configuration fails on the variables, the SM index and size are correct but all variables are 0. I used the following code:
`                ETHERCATSlaveInfo.Descendants().ToList()[0].DynamicData.PdoSet.Clear();

                //PDO Config OUTPUTS
                SlavePdo CMMTPdo_OUT = new SlavePdo(ETHERCATSlaveInfo, "Outputs", 0x1600, 0, true, true, 2); //SyncManager: 2-Saidas; 3-Entradas;
                List<SlaveVariable> ecCMMTVariableSet_OUT = new List<SlaveVariable>();
                ecCMMTVariableSet_OUT.Add(new SlaveVariable(CMMTPdo_OUT, "ControlWord", 0x6040, 0, DataDirection.Output, OneDasDataType.UINT16));
                ecCMMTVariableSet_OUT.Add(new SlaveVariable(CMMTPdo_OUT, "Modes of operation", 0x6060, 0, DataDirection.Output, OneDasDataType.INT8));
                ecCMMTVariableSet_OUT.Add(new SlaveVariable(CMMTPdo_OUT, "Target position", 0x607A, 0, DataDirection.Output, OneDasDataType.INT32));
                ecCMMTVariableSet_OUT.Add(new SlaveVariable(CMMTPdo_OUT, "Profile velocity", 0x6081, 0, DataDirection.Output, OneDasDataType.UINT32));
                ecCMMTVariableSet_OUT.Add(new SlaveVariable(CMMTPdo_OUT, "Target velocity", 0x60FF, 0, DataDirection.Output, OneDasDataType.INT32));
                ecCMMTVariableSet_OUT.Add(new SlaveVariable(CMMTPdo_OUT, "Target torque", 0x6071, 0, DataDirection.Output, OneDasDataType.INT16));
                ecCMMTVariableSet_OUT.Add(new SlaveVariable(CMMTPdo_OUT, "Velocity offset", 0x60B1, 0, DataDirection.Output, OneDasDataType.INT32));
                ecCMMTVariableSet_OUT.Add(new SlaveVariable(CMMTPdo_OUT, "Torque offset", 0x60B2, 0, DataDirection.Output, OneDasDataType.INT16));
                ecCMMTVariableSet_OUT.Add(new SlaveVariable(CMMTPdo_OUT, "", 0x0, 0, DataDirection.Output, OneDasDataType.BOOLEAN));
                CMMTPdo_OUT.SetVariableSet(ecCMMTVariableSet_OUT);
                ETHERCATSlaveInfo.Descendants().ToList()[0].DynamicData.PdoSet.Add(CMMTPdo_OUT);

                //PDO Config INPUTS
                SlavePdo CMMTPdo_IN = new SlavePdo(ETHERCATSlaveInfo, "Inputs", 0x1A00, 0, true, true, 3); //SyncManager: 2-Saidas; 3-Entradas;
                List<SlaveVariable> ecCMMTVariableSet_IN = new List<SlaveVariable>();
                ecCMMTVariableSet_IN.Add(new SlaveVariable(CMMTPdo_IN, "Statusword", 0x6041, 0, DataDirection.Input, OneDasDataType.UINT16));
                ecCMMTVariableSet_IN.Add(new SlaveVariable(CMMTPdo_IN, "Modes of operation display", 0x6061, 0, DataDirection.Input, OneDasDataType.INT8));
                ecCMMTVariableSet_IN.Add(new SlaveVariable(CMMTPdo_IN, "Position actual value", 0x6064, 0, DataDirection.Input, OneDasDataType.INT32));
                ecCMMTVariableSet_IN.Add(new SlaveVariable(CMMTPdo_IN, "Velocity actual value", 0x606C, 0, DataDirection.Input, OneDasDataType.INT32));
                ecCMMTVariableSet_IN.Add(new SlaveVariable(CMMTPdo_IN, "Torque actual value", 0x6077, 0, DataDirection.Input, OneDasDataType.INT16));
                ecCMMTVariableSet_IN.Add(new SlaveVariable(CMMTPdo_IN, "", 0x0, 0, DataDirection.Input, OneDasDataType.BOOLEAN));
                CMMTPdo_IN.SetVariableSet(ecCMMTVariableSet_IN);
                ETHERCATSlaveInfo.Descendants().ToList()[0].DynamicData.PdoSet.Add(CMMTPdo_IN);`

And the output from the slave software:

[image]

Apollo3zehn, 2020-05-27:
Could you please confirm the following changes:
- conversion of the product code from `8084005` to `#x7B5A25`
- conversion of `<SubIndex>0</SubIndex>` to `<SubIndex>#x0</SubIndex>`
- application of fix from #47
- replace `(byte)` by `(ushort)` to correctly write the PDO assignment
https://github.com/Apollo3zehn/EtherCAT.NET/blob/5ea922a0101636ed17a41563c958a7db5b34a032/src/EtherCAT.NET/Infrastructure/SlaveInfo.cs#L184
- replace `(byte)` by `(ushort)` to correctly write the PDO mapping
https://github.com/Apollo3zehn/EtherCAT.NET/blob/5ea922a0101636ed17a41563c958a7db5b34a032/src/EtherCAT.NET/Infrastructure/SlaveInfo.cs#L154

Thank you!

Regarding the new issue: Could you please attach another Wireshark log with your manual PDO configuration?

joaosbastos, 2020-05-27:
That is correct.

I attach the wireshark capture from the manual configuration with only the FESTO slave:

[Wireshark_ManualConfig.zip](https://github.com/Apollo3zehn/EtherCAT.NET/files/4688306/Wireshark_ManualConfig.zip)

I tried the modified code with another slave helmholz TB20C and it worked fine.
Then i added both slaves on the network and got the following error

`ConfigureDc failed (0x0302): The static drift compensation for distributed clocks failed. :    at EtherCAT.NET.EcUtilities.CheckErrorCode(IntPtr context, Int32 errorCode, String callerMemberName) in C:\Users\jbast\Downloads\EtherCAT.NET\src\EtherCAT.NET\EcUtilities.cs:line 148
   at EtherCAT.NET.EcMaster.ConfigureDc() in C:\Users\jbast\Downloads\EtherCAT.NET\src\EtherCAT.NET\EcMaster.cs:line 226
   at EtherCAT.NET.EcMaster.Configure(SlaveInfo rootSlaveInfo) in C:\Users\jbast\Downloads\EtherCAT.NET\src\EtherCAT.NET\EcMaster.cs:line 291`

I attach the wireshark for both slaves on the network and the ESI:

[Wireshark_2slaves.zip](https://github.com/Apollo3zehn/EtherCAT.NET/files/4688318/Wireshark_2slaves.zip)
[Helmholz TB20 Coupler EtherCAT.txt](https://github.com/Apollo3zehn/EtherCAT.NET/files/4688320/Helmholz.TB20.Coupler.EtherCAT.txt)

Apollo3zehn, 2020-05-27:
One issue is that you create your PDOs incorrectly: 
`new SlavePdo(null, ecPdoInfo.name, ecPdoInfo.index, 0, true, true, smIndex - 0x1C10);`
should be:
`new SlavePdo(null, ecPdoInfo.name, ecPdoInfo.index, 0, false, true, smIndex - 0x1C10);`

When the parameter `isFixed` is set to true, EtherCAT.NET will not write the PDO content to the slave because it is fixed and cannot be modifed. So set it to false, like it is in the ESI file (see `<TxPdo Fixed="0" Sm="3">`).

I will later look into the issue with both slaves.

Apollo3zehn, 2020-05-27:
The error `The static drift compensation for distributed clocks failed.` means that although the master sent 15.000 frames to the slaves during the startup phase, the clock difference to the reference clock is still > 100 ns. Either you increase the limit to send more frames (no guarantee that this works), or you increase the limit to > 100 ns. If you don't need distributed clocks, it is easier to increase the limit.

https://github.com/Apollo3zehn/EtherCAT.NET/blob/5ea922a0101636ed17a41563c958a7db5b34a032/src/EtherCAT.NET/EcSettings.cs#L10-L11

e.g.:
```cs
var settings = new EcSettings(cycleFrequency, esiDirectoryPath, interfaceName) 
{ 
    TargetTimeDifference = 1000 // 1000 ns
};
```

joaosbastos, 2020-05-27:
The TargetTimeDifference = 1000 solved the 2 slaves issue.

The manual configuration i changed to:
`                ETHERCATSlaveInfo.Descendants().ToList()[0].DynamicData.PdoSet.Clear();

                //PDO Config OUTPUTS
                SlavePdo CMMTPdo_OUT = new SlavePdo(ETHERCATSlaveInfo, "Outputs", 0x1600, 0, false, true, 2); //SyncManager: 2-Saidas; 3-Entradas; Fixed=True não escreve para o escravo
                List<SlaveVariable> ecCMMTVariableSet_OUT = new List<SlaveVariable>();
                ecCMMTVariableSet_OUT.Add(new SlaveVariable(CMMTPdo_OUT, "ControlWord", 0x6040, 0, DataDirection.Output, OneDasDataType.UINT16));
                ecCMMTVariableSet_OUT.Add(new SlaveVariable(CMMTPdo_OUT, "Modes of operation", 0x6060, 0, DataDirection.Output, OneDasDataType.INT8));
                ecCMMTVariableSet_OUT.Add(new SlaveVariable(CMMTPdo_OUT, "Target position", 0x607A, 0, DataDirection.Output, OneDasDataType.INT32));
                ecCMMTVariableSet_OUT.Add(new SlaveVariable(CMMTPdo_OUT, "Profile velocity", 0x6081, 0, DataDirection.Output, OneDasDataType.UINT32));
                ecCMMTVariableSet_OUT.Add(new SlaveVariable(CMMTPdo_OUT, "Target velocity", 0x60FF, 0, DataDirection.Output, OneDasDataType.INT32));
                ecCMMTVariableSet_OUT.Add(new SlaveVariable(CMMTPdo_OUT, "Target torque", 0x6071, 0, DataDirection.Output, OneDasDataType.INT16));
                ecCMMTVariableSet_OUT.Add(new SlaveVariable(CMMTPdo_OUT, "Velocity offset", 0x60B1, 0, DataDirection.Output, OneDasDataType.INT32));
                ecCMMTVariableSet_OUT.Add(new SlaveVariable(CMMTPdo_OUT, "Torque offset", 0x60B2, 0, DataDirection.Output, OneDasDataType.INT16));
                ecCMMTVariableSet_OUT.Add(new SlaveVariable(CMMTPdo_OUT, "", 0x0, 0, DataDirection.Output, OneDasDataType.BOOLEAN));
                CMMTPdo_OUT.SetVariableSet(ecCMMTVariableSet_OUT);
                ETHERCATSlaveInfo.Descendants().ToList()[0].DynamicData.PdoSet.Add(CMMTPdo_OUT);

                //PDO Config INPUTS
                SlavePdo CMMTPdo_IN = new SlavePdo(ETHERCATSlaveInfo, "Inputs", 0x1A00, 0, false, true, 3); //SyncManager: 2-Saidas; 3-Entradas; Fixed=True não escreve para o escravo
                List<SlaveVariable> ecCMMTVariableSet_IN = new List<SlaveVariable>();
                ecCMMTVariableSet_IN.Add(new SlaveVariable(CMMTPdo_IN, "Statusword", 0x6041, 0, DataDirection.Input, OneDasDataType.UINT16));
                ecCMMTVariableSet_IN.Add(new SlaveVariable(CMMTPdo_IN, "Modes of operation display", 0x6061, 0, DataDirection.Input, OneDasDataType.INT8));
                ecCMMTVariableSet_IN.Add(new SlaveVariable(CMMTPdo_IN, "Position actual value", 0x6064, 0, DataDirection.Input, OneDasDataType.INT32));
                ecCMMTVariableSet_IN.Add(new SlaveVariable(CMMTPdo_IN, "Velocity actual value", 0x606C, 0, DataDirection.Input, OneDasDataType.INT32));
                ecCMMTVariableSet_IN.Add(new SlaveVariable(CMMTPdo_IN, "Torque actual value", 0x6077, 0, DataDirection.Input, OneDasDataType.INT16));
                ecCMMTVariableSet_IN.Add(new SlaveVariable(CMMTPdo_IN, "", 0x0, 0, DataDirection.Input, OneDasDataType.BOOLEAN));
                CMMTPdo_IN.SetVariableSet(ecCMMTVariableSet_IN);
                ETHERCATSlaveInfo.Descendants().ToList()[0].DynamicData.PdoSet.Add(CMMTPdo_IN);`

It still doesn't go to RUN the wireshark capture:

[Wireshark_ManualConfig_FixedFalse.zip](https://github.com/Apollo3zehn/EtherCAT.NET/files/4689015/Wireshark_ManualConfig_FixedFalse.zip)

Is there a way to leave the slave from RUN?

joaosbastos, 2020-05-27:
If i fill in the bitlength it works

`                ETHERCATSlaveInfo.Descendants().ToList()[0].DynamicData.PdoSet.Clear();

                //PDO Config OUTPUTS
                SlavePdo CMMTPdo_OUT = new SlavePdo(ETHERCATSlaveInfo, "Outputs", 0x1600, 0, false, true, 2); //SyncManager: 2-Saidas; 3-Entradas; Fixed=True não escreve para o escravo
                List<SlaveVariable> ecCMMTVariableSet_OUT = new List<SlaveVariable>();
                ecCMMTVariableSet_OUT.Add(new SlaveVariable(CMMTPdo_OUT, "ControlWord", 0x6040, 0, DataDirection.Output, OneDasDataType.UINT16, 16));
                ecCMMTVariableSet_OUT.Add(new SlaveVariable(CMMTPdo_OUT, "Modes of operation", 0x6060, 0, DataDirection.Output, OneDasDataType.INT8, 8));
                ecCMMTVariableSet_OUT.Add(new SlaveVariable(CMMTPdo_OUT, "Target position", 0x607A, 0, DataDirection.Output, OneDasDataType.INT32, 32));
                ecCMMTVariableSet_OUT.Add(new SlaveVariable(CMMTPdo_OUT, "Profile velocity", 0x6081, 0, DataDirection.Output, OneDasDataType.UINT32, 32));
                ecCMMTVariableSet_OUT.Add(new SlaveVariable(CMMTPdo_OUT, "Target velocity", 0x60FF, 0, DataDirection.Output, OneDasDataType.INT32, 32));
                ecCMMTVariableSet_OUT.Add(new SlaveVariable(CMMTPdo_OUT, "Target torque", 0x6071, 0, DataDirection.Output, OneDasDataType.INT16, 16));
                ecCMMTVariableSet_OUT.Add(new SlaveVariable(CMMTPdo_OUT, "Velocity offset", 0x60B1, 0, DataDirection.Output, OneDasDataType.INT32, 32));
                ecCMMTVariableSet_OUT.Add(new SlaveVariable(CMMTPdo_OUT, "Torque offset", 0x60B2, 0, DataDirection.Output, OneDasDataType.INT16, 16));
                ecCMMTVariableSet_OUT.Add(new SlaveVariable(CMMTPdo_OUT, "", 0x0, 0, DataDirection.Output, OneDasDataType.BOOLEAN, 8));
                CMMTPdo_OUT.SetVariableSet(ecCMMTVariableSet_OUT);
                ETHERCATSlaveInfo.Descendants().ToList()[0].DynamicData.PdoSet.Add(CMMTPdo_OUT);

                //PDO Config INPUTS
                SlavePdo CMMTPdo_IN = new SlavePdo(ETHERCATSlaveInfo, "Inputs", 0x1A00, 0, false, true, 3); //SyncManager: 2-Saidas; 3-Entradas; Fixed=True não escreve para o escravo
                List<SlaveVariable> ecCMMTVariableSet_IN = new List<SlaveVariable>();
                ecCMMTVariableSet_IN.Add(new SlaveVariable(CMMTPdo_IN, "Statusword", 0x6041, 0, DataDirection.Input, OneDasDataType.UINT16, 16));
                ecCMMTVariableSet_IN.Add(new SlaveVariable(CMMTPdo_IN, "Modes of operation display", 0x6061, 0, DataDirection.Input, OneDasDataType.INT8, 8));
                ecCMMTVariableSet_IN.Add(new SlaveVariable(CMMTPdo_IN, "Position actual value", 0x6064, 0, DataDirection.Input, OneDasDataType.INT32, 32));
                ecCMMTVariableSet_IN.Add(new SlaveVariable(CMMTPdo_IN, "Velocity actual value", 0x606C, 0, DataDirection.Input, OneDasDataType.INT32, 32));
                ecCMMTVariableSet_IN.Add(new SlaveVariable(CMMTPdo_IN, "Torque actual value", 0x6077, 0, DataDirection.Input, OneDasDataType.INT16, 16));
                ecCMMTVariableSet_IN.Add(new SlaveVariable(CMMTPdo_IN, "", 0x0, 0, DataDirection.Input, OneDasDataType.BOOLEAN, 8));
                CMMTPdo_IN.SetVariableSet(ecCMMTVariableSet_IN);
                ETHERCATSlaveInfo.Descendants().ToList()[0].DynamicData.PdoSet.Add(CMMTPdo_IN);`

Only missing the way to take the slave from RUN.

Apollo3zehn, 2020-05-27:
Do you mean not starting the slave (OP) and let it be at SAFE-OP? If yes, there is no such option currently. Although you can modify the code 
https://github.com/Apollo3zehn/EtherCAT.NET/blob/5ea922a0101636ed17a41563c958a7db5b34a032/src/EtherCAT.NET/EcMaster.cs#L302-L306

And also modify the native code:

https://github.com/Apollo3zehn/EtherCAT.NET/blob/5ea922a0101636ed17a41563c958a7db5b34a032/native/SOEM_wrapper/soem_wrapper.c#L332-L351

But the native part is more related to the underlying SOEM library. E.g. `context->slavelist[0].state = EC_STATE_OPERATIONAL;` means that all slaves shall go into OP (`slavelist[0]` denotes the master itself).

Regarding the bit lengths: yes you are right, that is one option. The other would be to replace `OneDasDataType.BOOLEAN` by `OneDasDataType.UINT8`.

The first issue is that BOOLEAN is not the correct data type because the length is 1 bit but you need 8 (OneDasDataType.UINT8). But apart from that, the master sends an `0x08` instead of `0x01` in case of register `0x1600` and `0x06` instead of `0x01` in case of register `0x1600`. I do not know yet why this is happening. In any case, switching to OneDasDataType.UINT8 should also solve the issue.

Apollo3zehn, 2020-05-27:
For what purpose do you need to switch back from OP to SAFE-OP?

joaosbastos, 2020-05-27:
To recover from communication error, so i can restart the network.

joaosbastos, 2020-05-27:
Also my outputs get freezed.

Apollo3zehn, 2020-05-27:
### Recovering

There is a watchdog task running in parallel that should reconfigure all slaves when a slave has an error or for example the network cable is pulled (and plugged in back):
https://github.com/Apollo3zehn/EtherCAT.NET/blob/master/src/EtherCAT.NET/EcMaster.cs#L409-L459

If you want to recover a single slave instead, you may call this P/Invoke method:

https://github.com/Apollo3zehn/EtherCAT.NET/blob/1ecc3c751736e9526543b0bd8415e3c0597aed3a/src/SOEM.PInvoke/PInvoke/EcConfig.cs#L21-L23

e.g.
`EcConfig.ecx_recover_slave(ecMaster.Context, slaveNumber, timeout);`

Howevery, if I remember correctly, this will not reconfigure the PDO assignment and PDO mapping, so it may not work.

### Output Freezing

Output freezing may happen due to a process data watchdog timeout on the slave (see [Hardware Data SheetSection I](https://download.beckhoff.com/download/document/io/ethercat-development-products/ethercat_esc_datasheet_sec1_technology_2i2.pdf) from Beckhoff):

[image]

This master is not real-time capable so the watchdog is disabled here: https://github.com/Apollo3zehn/EtherCAT.NET/blob/master/native/SOEM_wrapper/soem_wrapper.c#L561-L572

So I would expect that outputs cannot freeze. I have no explanation why they are still freezing.

joaosbastos, 2020-05-27:
**Output Freezing**
That explains it, when the slave goes offline without the watchdog the outputs keep their last state.

**Recovering**
I understand now.

Thank you.

Apollo3zehn, 2020-05-28:
I got an answer in the ETG forum with a link to the ETG.1020 spec Clause 12.2, sub-point 1 "SDO Complete Access". There it is written that when using Complete Access to transmit the CoE data - which is the case with your slave - the subindex 0 (i.e. number of entries) is **padded to 16 bits**. 

That explains why sometimes `byte` is sufficient and sometimes I need to cast the value to `ushort`: With complete access, we need `ushort`, without we need `byte`. I will publish an update today trying to fix the list of problems (https://github.com/Apollo3zehn/EtherCAT.NET/issues/49#issuecomment-634581538)

Apollo3zehn, 2020-05-28:
The master branch is now updated with a few fixes for the things we collected here. I´ll close this issue for now as the main issues are solved. Maybe you can test the changes? It should not be required anymore to modify the ESI file. Unfortunately I have no hardware available for tests and will be very busy the next two weeks.

closed 2020-05-28 by Apollo3zehn

joaosbastos, 2020-05-28:
I'll test it tomorrow and post here the result.
Thank you.

joaosbastos, 2020-05-29:
I tested with the ESI without modifications and got the error:

`Value cannot be null.
Parameter name: String :    at System.Number.StringToNumber(String str, NumberStyles options, NumberBuffer& number, NumberFormatInfo info, Boolean parseDecimal)
   at System.Number.ParseInt32(String s, NumberStyles style, NumberFormatInfo info)
   at System.Byte.Parse(String s, NumberStyles style, NumberFormatInfo info)
   at EtherCAT.NET.Extensibility.ExtensibilityHelper.<>c__DisplayClass0_1.<CreateDynamicData>b__1(PdoTypeEntry x) in C:\projects\ethercat-net\src\EtherCAT.NET\Extensibility\ExtensibilityHelper.cs:line 82
   at System.Linq.Enumerable.WhereSelectArrayIterator`2.MoveNext()
   at System.Collections.Generic.List`1..ctor(IEnumerable`1 collection)
   at System.Linq.Enumerable.ToList[TSource](IEnumerable`1 source)
   at EtherCAT.NET.Extensibility.ExtensibilityHelper.CreateDynamicData(String esiDirectoryPath, IExtensionFactory extensionFactory, SlaveInfo slaveInfo) in C:\projects\ethercat-net\src\EtherCAT.NET\Extensibility\ExtensibilityHelper.cs:line 88`

Apollo3zehn, 2020-05-29:
Thanks for your test. Sorry that I overlooked that, I made a change to the code that I reverted now (https://github.com/Apollo3zehn/EtherCAT.NET/commit/2105724ef805fe6b0797ef9b518e14738ab439d7). Is it working with this fix? I tested it with your ESI file but I can only test some of the code because I have no real slave available.

joaosbastos, 2020-06-05:
Sorry for the delay re-testing. I still got the same error:
`Value cannot be null.
Parameter name: String :    at System.Number.StringToNumber(String str, NumberStyles options, NumberBuffer& number, NumberFormatInfo info, Boolean parseDecimal)
   at System.Number.ParseInt32(String s, NumberStyles style, NumberFormatInfo info)
   at System.Byte.Parse(String s, NumberStyles style, NumberFormatInfo info)
   at EtherCAT.NET.Extensibility.ExtensibilityHelper.<>c__DisplayClass0_1.<CreateDynamicData>b__1(PdoTypeEntry x) in C:\projects\ethercat-net\src\EtherCAT.NET\Extensibility\ExtensibilityHelper.cs:line 82
   at System.Linq.Enumerable.WhereSelectArrayIterator`2.MoveNext()
   at System.Collections.Generic.List`1..ctor(IEnumerable`1 collection)
   at System.Linq.Enumerable.ToList[TSource](IEnumerable`1 source)
   at EtherCAT.NET.Extensibility.ExtensibilityHelper.CreateDynamicData(String esiDirectoryPath, IExtensionFactory extensionFactory, SlaveInfo slaveInfo) in C:\projects\ethercat-net\src\EtherCAT.NET\Extensibility\ExtensibilityHelper.cs:line 88`

I'm using the following ESI's:
[Festo-CMMT-ST-CiA402-20200113.txt](https://github.com/Apollo3zehn/EtherCAT.NET/files/4735002/Festo-CMMT-ST-CiA402-20200113.txt)
[Helmholz TB20 Coupler EtherCAT.txt](https://github.com/Apollo3zehn/EtherCAT.NET/files/4735003/Helmholz.TB20.Coupler.EtherCAT.txt)