XzuluX/EtherCAT.NET #54: fail to compile

Federau, 2021-02-07:
I tried to compile c lib in windows7 x64,but failed.
when I run init_solution.ps1 
powershell shows  the source directory D:/zqFile/EtherCat/Ethercat.net-master/native/soem dose not contain a cmakelists.txt file

logfile shows following error:
 cl /c /Zi /W3 /WX- /diagnostics:column /Od /Ob0 /Oy- /D WIN32 /D _WINDOWS /D "CMAKE_INTDIR=\"Debug\"" /D _MBCS /Gm- /EHsc /RTC1 /MDd /GS /fp:precise /Zc:wchar_t /Zc:forScope /Zc:inline /GR /Fo"cmTC_61f26.dir\Debug\\" /Fd"cmTC_61f26.dir\Debug\vc142.pdb" /Gd /TP /analyze- /errorReport:queue "C:\Program Files\CMake\share\cmake-3.19\Modules\CMakeCXXCompilerABI.cpp"  CMakeCXXCompilerABI.cpp

Apollo3zehn, 2021-02-07:
Is the /nativ/soem/ submodule correctly initialized? I.e. is the cmakelists file there and the rest of soem? It should look like https://github.com/OpenEtherCATsociety/SOEM.

Federau, 2021-02-07:
Actrually, as u mentioned above,  it occured a mistake while initializing submoudule.maybe there are some problems with my git, I just commented out the git command from  powershell script, and  cloned source code from https://github.com/OpenEtherCATsociety/SOEM. It really works! Then I try to compile win32 project(SoemWrapper), there is an error in soem_wrapper.c  line 119 . struct ecx_context does not contains field DCt0 and DCl. maybe there is a change in this struct definition in soem source code?

Apollo3zehn, 2021-02-07:
Try to checkout release 1.4.0: https://github.com/OpenEtherCATsociety/SOEM/tree/v1.4.0

It could be that there habe been changes in the meantime.

Federau, 2021-02-07:
All projects compiled successfully, Thank you for your timely reply and  selfless contribution.Thank u  so much!

closed 2021-02-07 by Federau