XzuluX/EtherCAT.NET #55: AL Status Code 0x0050 - EEPROM No Access

yschmid-levi, 2021-02-10:
Hi,

I was playing around with the sample application and run into an issue with EtherCAT slaves that need access to the SII EEPROM during the INIT->PREOP transition. The slaves are stuck in Init+Error with AL Status Code 0x0050 (EEPROM No Access).

According to ETG.1020 *The EtherCAT master shall give the EEPROM access rights to the application controller [...] in the transition INIT-PREOP*. 

The issue seems to be in `ScanDevices()`:
 - in `ecx_config_init ()` PDI asscess to the EEPROM is granted before requesting PREOP: https://github.com/OpenEtherCATsociety/SOEM/blob/d9261e801d109eaf99b94907b6d24bac4a123968/soem/ethercatconfig.c#L590-L601
 - but in `ScanDevices()` access is immediatly taken by the master to access the Configured Station Alias(CSA): https://github.com/Apollo3zehn/EtherCAT.NET/blob/ba846c525290bb4ea8273659cff8c464c1a79ce7/native/SOEM_wrapper/soem_wrapper.c#L554-L567

Waiting for the slaves to reach PREOP before accessing the CSA in EEPROM solves this issue.

Cheers,
Yves

Apollo3zehn, 2021-02-10:
Thanks! Probably I will have some time to look into this issue on friday.

closed 2021-02-16 by Apollo3zehn

Apollo3zehn, 2021-02-16:
I hope its working better now, I have added the following lines: https://github.com/Apollo3zehn/EtherCAT.NET/blob/master/native/SOEM_wrapper/soem_wrapper.c#L562-L566

Could you please confirm, if this solves your issue? There is a new version on nuget if you need it: https://www.nuget.org/packages/EtherCAT.NET/1.0.0-alpha.5.final

yschmid-levi, 2021-02-16:
It works 👍 . I would probably return an error instead of looping for eternity if a slave is stuck in INIT.

commit 9ad66bb8 (2021-02-16): Wait for PREOP before accessing EEPROM (fixes #55).)