XzuluX/EtherCAT.NET #61: where is the SOEM_wrapper folder?

Jacklove-G, 2021-04-20:
After  ./init_solution.ps1 step, and when I run  msbuild ./artifacts/bin32/SOEM_wrapper/soem_wrapper.vcxproj, there is no folder called SOEM_wrapper

Jacklove-G, 2021-04-20:
when I run ./init_solution.ps1 with $true, and something wrong happens

CMake Error at CMakeLists.txt:11 (add_subdirectory):
  The source directory

    C:/Users/Administrator/Desktop/EtherCAT.NET-master/EtherCAT.NET-master/native/SOEM

  does not contain a CMakeLists.txt file.

Apollo3zehn, 2021-04-20:
What happens if you run it again with a clean repo and with the parameter set to $true? I thin the error you see appears when the initial run oft the powershell script failed.

closed 2021-07-06 by Apollo3zehn