XzuluX/EtherCAT.NET #62: No CMakeLists.txt in folder SOEM of EtherCAT.NET-master

Bit2Bi, 2021-05-02:
Hi,
when building EtherCAT.NET-master with powerScript .\init_solution.ps1  the process ended with error.
It says: „... does not contain a CMAkeLists.txt file“
Obiously there is no such file in the folder .\..\native\SOEM.
Consequently no soem_wrapper.dll is being built.

I am using VS2019, powerscript 7.x, net core 3.1 etc.
How can I solve the issue.
Regards

Apollo3zehn, 2021-05-02:
What happens if you make a clean `git clone` and then run the script? What is the console output?

Bit2Bi, 2021-05-02:
Well, I downloaded the code as .zip and opened it with 7z. 
First thing when run in PS is: term ‘git’ is not recognized. I have no clue why this happens!
Second issue is as described above; no CMakeLists.txt where found in the ./../native/ SOEM.

Maybe anything wrong with PS settings?

Apollo3zehn, 2021-05-03:
Sorry I presumed you are using git to clone the reposity. Downloading the .zip and extracting it should make no difference.

Do you have Visual Studio with the C++ workload installed? Thiss workload provides the C++ compiler required for cmake.

[image]

The easiest for me to find the reason would be if you take a clean copy of EtherCAT.NET, run the powershell script and post the full output here.

Here is the expeced output

```pwsh
PS C:\Users\wilvin\Desktop\Test\Neuer Ordner> git clone https://github.com/Apollo3zehn/EtherCAT.NET
Cloning into 'EtherCAT.NET'...
remote: Enumerating objects: 630, done.
remote: Counting objects: 100% (331/331), done.
remote: Compressing objects: 100% (224/224), done.

Receiving objects: 100% (630/630), 221.08 KiB | 1.64 MiB/s, done.
Resolving deltas: 100% (318/318), done.
PS C:\Users\wilvin\Desktop\Test\Neuer Ordner> cd .\EtherCAT.NET\
PS C:\Users\wilvin\Desktop\Test\Neuer Ordner\EtherCAT.NET> .\init_solution.ps1
Updating Git submodule.
Creating native x86 project.

    Directory: C:\Users\wilvin\Desktop\Test\Neuer Ordner\EtherCAT.NET\artifacts

Mode                 LastWriteTime         Length Name
----                 -------------         ------ ----
d----          03.05.2021    10:27                bin32
-- The C compiler identification is MSVC 19.28.29913.0
-- The CXX compiler identification is MSVC 19.28.29913.0
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.28.29910/bin/Hostx64/x86/cl.exe
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.28.29910/bin/Hostx64/x86/cl.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.28.29910/bin/Hostx64/x86/cl.exe
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.28.29910/bin/Hostx64/x86/cl.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
OS is win32
LIB_DIR: lib
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/wilvin/Desktop/Test/Neuer Ordner/EtherCAT.NET/artifacts/bin32
Creating native x64 project.
d----          03.05.2021    10:27                bin64
-- The C compiler identification is MSVC 19.28.29913.0
-- The CXX compiler identification is MSVC 19.28.29913.0
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.28.29910/bin/Hostx64/x64/cl.exe
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.28.29910/bin/Hostx64/x64/cl.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.28.29910/bin/Hostx64/x64/cl.exe
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.28.29910/bin/Hostx64/x64/cl.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
OS is win32
LIB_DIR: lib
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/wilvin/Desktop/Test/Neuer Ordner/EtherCAT.NET/artifacts/bin64
```

Bit2Bi, 2021-05-03:
Hi,
thanks for advice. After reinstalling the git and adjusting the environmental var’s it finally worked. I can see the dll’s for the wrapper. However, they are not created when running the EtherCAT.net.sln in the VS2019 IDE. 
I keep struggling 
Best wishes

Apollo3zehn, 2021-05-03:
The native DLLs are not created when compiling the managed DLLs (EtherCAT.NET.sln). They can be compiled using msbuild:
```
msbuild ./artifacts/bin32/SOEM_wrapper/soem_wrapper.vcxproj /p:Configuration=Release
msbuild ./artifacts/bin64/SOEM_wrapper/soem_wrapper.vcxproj /p:Configuration=Release
```

I have no test setup right now, but somewhere in the artifacs/bin32 or bin64 folder, there should be another .sln file which can generate the native DLLs.

Bit2Bi, 2021-05-05:
Hi,
meanwhile the code compiles fine without error and can run. But, excecution failed at EcUtilities here:         EcUtilities.CheckErrorCode(context, EcHL.ScanDevices(context, interfaceName, out var slaveIdentifications, out var slaveCount));
error 0x104, did not reach PreOp-State.

I can run slaveinfo.exe on same system with success.
Anything overlooked?

Apollo3zehn, 2021-05-06:
Are you able to read out the slave state, right after catching the exception?

Maybe calling `Ecutilities.GetSlaveStateDescription(IntPtr context, IEnumerable<SlaveInfo> slaves)` helps to find the reason for the slave not getting into the PreOp-State.

```cs
var master = xxxx;
Ecutilities.GetSlaveStateDescription(master.Context, slaves);
```

The important part of the resulting string is the 'AL-Status' which can be translated [here](https://infosys.beckhoff.de/index.php?content=../content/1031/ethercatsystem/1037010571.html&id=) or [here](https://github.com/OpenEtherCATsociety/SOEM/blob/f9db4f2b23455e83d9871568e3a69ab124e0c1a0/soem/ethercatprint.c#L97-L150).

Bit2Bi, 2021-05-06:
I get no slaveinfo yet. The system collapses before retrieving the slaves so as the status code.
The output on exception is:
An unhandled exception of type 'System.Exception' occurred in System.Private.CoreLib.dll
ScanDevices failed (0x0104): One or more terminals did not reach PreOp-state.
hm!