XzuluX/EtherCAT.NET #63: compile source code error

CN-BR-Soft, 2021-06-04:
**Hi,
 there is error happened when  i run .\init_solution.ps1 this command. would you like to tell me why? thanks.**

PS D:\Ethercat\EtherCAT.NET> .\init_solution.ps1
Updating Git submodule.
Creating native x86 project.

    目录: D:\Ethercat\EtherCAT.NET\artifacts

Mode                 LastWriteTime         Length Name
----                 -------------         ------ ----
d-----   2021/6/4 星期五     16:37                bin32
System.PlatformNotSupportedException
所在位置 D:\Ethercat\EtherCAT.NET\init_solution.ps1:25 字符: 5
+     throw [System.PlatformNotSupportedException]
+     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : OperationStopped: (System.PlatformNotSupportedException:RuntimeType) [], RuntimeExceptio
   n
    + FullyQualifiedErrorId : System.PlatformNotSupportedException

OS: win10 64bit
VS2019
.net core3.1

CN-BR-Soft, 2021-06-04:
[image]

Apollo3zehn, 2021-06-05:
Hi, this looks strange. I will have a look into this in Monday.

Apollo3zehn, 2021-06-08:
You need to use Powershell Core (i.e. Powershell 7), or edit the script and set `$IsWindows = $true`.

closed 2021-07-06 by Apollo3zehn