XzuluX/EtherCAT.NET #64: Help Needed

klchin66, 2021-06-15:
Hi,

Under "soem_wrapper.c"

int CALLCONV ScanDevices(ecx_contextt* context, char* interfaceName, ec_slave_info_t** slaveInfoSet, int* slaveCount)

        // request PREOP state for all slaves
        int counter = 5;

        do
        {
            ecx_statecheck(context, 0, EC_STATE_PRE_OP, EC_TIMEOUTSTATE);
        } while (counter-- && (context->slavelist[0].state != (EC_STATE_PRE_OP | EC_STATE_ACK)));

        if (context->slavelist[0].state != (EC_STATE_PRE_OP | EC_STATE_ACK))
            return -0x0104;

Should the Slave module required reply with EC_STATE_ACK?

But the "EC_STATE_ACK" was refer "Error or ACK error".  

Or the code should be 

        {
            ecx_statecheck(context, 0, EC_STATE_PRE_OP, EC_TIMEOUTSTATE);
        } while (counter-- && (context->slavelist[0].state != (EC_STATE_PRE_OP)));

        if (context->slavelist[0].state != (EC_STATE_PRE_OP))
            return -0x0104;

Regards,
KL Chin

Apollo3zehn, 2021-06-15:
Hi, a few month ago we had the same discussion. Please see my conclusion here:
https://github.com/Apollo3zehn/EtherCAT.NET/issues/57#issuecomment-784869641

Does your slave not return the ACK flag?

klchin66, 2021-06-15:
The salve module did not reply EC_STATE_ACK.
Tested with 1 or 2 salve, problem were same, the code never work for me.
My both salve using LAN9252 and AX58100 chip with stm32F407 (from China).

Apollo3zehn, 2021-06-16:
Can you debug into that code and see what the actual value of `context->slavelist[0].state` is? Or did you do that already? 
Does changing the code to the version you suggested work for your slaves?

klchin66, 2021-06-16:
a) '0' before entering to the loop
b) '2' immediate after ecx_statecheck -> ecx_BRD -> ecx_srconfirm was return '2'

My modified code works for my slave.  Not sure why so many DC drifted.

[image]

Apollo3zehn, 2021-06-18:
Thanks for your investigations. Unfortunately I have no slaves available to reproduce the behavior. I would like to keep this issue open and wait for others to report similar problems. Only then I can see if this behavior is specific to your slaves or a general behavior. With Beckhoff slaves I have to take the `EC_STATE_ACK` flag into account.

Many DC drifts may occur often when not operating on a real time system.

klchin66, 2021-06-18:
Will it better to by code below, so it cover both condition

            counter--;
          } while ( (0 < counter) && (EC_STATE_NONE == (context->slavelist[0].state) & (EC_STATE_PRE_OP | EC_STATE_ACK)) );
        if ( EC_STATE_NONE == (context->slavelist[0].state & (EC_STATE_PRE_OP | EC_STATE_PRE_OP)) )
          return ( -0x0104 );

From the sample salve code (SOES), I don't see it reply together with  EC_STATE_ACK.

klchin66, 2021-06-22:
From the WireShark, I got these result

[image]

[image]

[image]

Look likely it did receive STATE_ACK on the ecx_FPWRw( context->port,configadr,ECT_REG_ALCTL,htoes(EC_STATE_PRE_OP | EC_STATE_ACK),EC_TIMEOUTRET3 );

But not during ecx_statecheck( context,0,EC_STATE_PRE_OP,EC_TIMEOUTSTATE );

Anything wrong?

Apollo3zehn, 2021-07-06:
I am not sure, I will have to check this behavior against the EtherCAT specs. I am sorry, but unfortunately I have not much time right now to investigate this deeper. I will keep this issue open and come back when I have more time available.

jacckk, 2021-09-09:
I believe I'm having a similar issue. 

I'm using a HBM TIM EC EtherCAT module and with version EtherCAT.NET version 1.0.0-alpha.9.final. I get the error "ScanDevices failed (0x0104): One or more terminals did not reach PreOp-state.

Changing the check to ignore the ACK state allows it to work:
`
if (context->slavelist[0].state != (EC_STATE_PRE_OP))
       return -0x0104;
`

xutingjie168, 2021-09-16:
he funtion ( var rootSlave = EcUtilities.ScanDevices(context, interfaceName, referenceRootSlave);)but An error occurred "System.Exception:“ScanDevices failed (0x0104): One or more terminals did not reach PreOp-state."ignore the ACK state allows it to work,but can't find slaves  ,How to solve the issues

klchin66, 2021-09-16:
I changed two modules to resolve my problem.
"EC_STATE_ACK" was acknowledge by the salve during ecx_config_init,  but not ScanDevices.

a) ethercatconfig.c  
     -> int ecx_config_init( ecx_contextt *context,  uint8 usetable )

            //  some slaves need eeprom available to PDI in init->preop transition
            ecx_eeprom2pdi( context,slave );
            //  User may override automatic state change
            if ( 0 == context->manualstatechange )
              {
                //  request pre_op for slave - set preop status
                if ( 0 != ecx_FPWRw(context->port,configadr,ECT_REG_ALCTL,htoes(EC_STATE_PRE_OP | EC_STATE_ACK),EC_TIMEOUTRET3) )
                  irack++;
                EC_PRINT( "Slave %d receive STATE_ACK - %d.\n",slave,irack );
              }
            else
              {
                irack++;
              }
          }
      }
    return ( (irack << 16) | wkc );
}

b) soem_wrapper.cs  ->
    ScanDevices( ecx_contextt *context,  char *interfaceName,  ec_slave_info_t **slaveInfoSet,  int *slaveCount )

        irack = ecx_config_init( context,false );
        wkc   = (irack & 0xFFFF);
        if ( 0 > wkc )
          {
            return ( -0x0103 );
          }
        irack         = (irack >> 16) & 0xFFFF;
        *slaveInfoSet = (ec_slave_info_t *)calloc( (size_t)(*context->slavecount + 1),sizeof(ec_slave_info_t) );
        *slaveCount   = *context->slavecount;
        counter = 5;
        do
          {
            counter--;
            ecx_statecheck( context,0,EC_STATE_PRE_OP,EC_TIMEOUTSTATE );
          //  } while ( (0 < counter) && ((EC_STATE_PRE_OP | EC_STATE_ACK) != context->slavelist[0].state) );
          } while ( (0 < counter) && (EC_STATE_NONE == (context->slavelist[0].state & (EC_STATE_PRE_OP | EC_STATE_ACK))) );
        if ( irack != *slaveCount )
          {
            //  if ( context->slavelist[0].state != (EC_STATE_PRE_OP | EC_STATE_ACK) )
            if ( EC_STATE_NONE == (context->slavelist[0].state & (EC_STATE_PRE_OP | EC_STATE_ACK)) )
              return ( -0x0104 );
          }

ga5tan, 2023-05-05:
I am also getting "One or more terminals did not reach PreOp-state" error with HNC servo using SamleMaster

ga5tan, 2023-05-05:
I did what @jacckk suggested (commenting error 104 out)
`if (context->slavelist[0].state != (EC_STATE_PRE_OP)) return -0x0104;`
and got little further. Now I got this error:

> Could not find ESI information of manufacturer '0x88' for slave with product code '0x88710000' and revision '0x1'

any ideas what to do next would be appreciated.

Apollo3zehn, 2023-05-05:
@ga5tan 

You need provide the ESI file for your device to EtherCAT.NET. You get this from the manufacturer. Then provide the path to your ESI folder as shown here: https://github.com/Apollo3zehn/EtherCAT.NET/blob/master/sample/SampleMaster/Program.cs#L48

Apollo3zehn, 2023-05-05:
Yes, ESI files are XML files, so it could work.

ga5tan, 2023-05-09:
It seems this code:
`var subIndex = Convert.ToByte(x.SubIndex);`
does not like value '#x00' as subindex.

XML looks like this (assuming that's the part, cause there is couple of such Index and SubIndex combinations.

Is #x00 no good as SubIndex? because most of the numerical values seem to have #x prefix

> <Index>#x603F</Index>
> <SubIndex>#x00</SubIndex>

ga5tan, 2023-05-09:
It seems to be an extension. 
I thought I try to make Sample work, but 'All I really need' is to able to command grid of servos to go to  specific position.
[HSD7_ES_EtherCAT_V120.zip](https://github.com/Apollo3zehn/EtherCAT.NET/files/11432081/HSD7_ES_EtherCAT_V120.zip)
attached smaller file

ga5tan, 2023-05-11:
Well, it almost looks like it working except this

`info: EtherCAT Master[0]
      Found 1 slaves:
      i7DS-**A30 (PDOs: 8 - CSA: 47163)
info: EtherCAT Master[0]
      IO map configured (1 slave, 32 bytes)
info: EtherCAT Master[0]
      DC system time diff. is <= 0 ns
fail: EtherCAT Master[0]
      CheckSafeOpState failed (0x0501): One or more terminals did not reach SafeOp-state.

Application failed with unhandled exception:
----------------------------------------------
System.Exception: CheckSafeOpState failed (0x0501): One or more terminals did not reach SafeOp-state.
   at EtherCAT.NET.EcUtilities.CheckErrorCode(IntPtr context, Int32 errorCode, String callerMemberName) in D:\dev\EtherCAT.NET\src\EtherCAT.NET\EcUtilities.cs:line 135
   at EtherCAT.NET.EcMaster.Configure(SlaveInfo rootSlave) in D:\dev\EtherCAT.NET\src\EtherCAT.NET\EcMaster.cs:line 277
   at SampleMaster.Program.Main(String[] args) in D:\dev\EtherCAT.NET\sample\SampleMaster\Program.cs:line 116
   at SampleMaster.Program.<Main>(String[] args)`

would it be safe/advisable to comment this line?

`EcUtilities.CheckErrorCode(this.Context, EcHL.CheckSafeOpState(this.Context), nameof(EcHL.CheckSafeOpState));`

Apollo3zehn, 2023-05-11:
That is the point where you would need to read out the status code of the slave:

https://github.com/Apollo3zehn/EtherCAT.NET/blob/8f60d91c220909e030ac77a27edd6411d5abadf2/src/SOEM.PInvoke/PInvoke/EcHL.cs#L341-L355

The `alStatusCode` tells you what is wrong. You can translate that code into an error description here:

https://infosys.beckhoff.com/index.php?content=../content/1031/ethercatsystem/1037010571.html&id=

ga5tan, 2023-05-12:
@Apollo3zehn weirdest thing. I added this
`Console.WriteLine(EcUtilities.GetSlaveStateDescription(master.Context, slaves.SelectMany(x => x.Descendants()).ToList()));`
but no error today.

Program just sits there with this output.
`Slaves: 1
info: EtherCAT Master[0]
      Found 1 slaves:
      i7DS-**A30 (PDOs: 8 - CSA: 47163)
info: EtherCAT Master[0]
      IO map configured (1 slave, 32 bytes)
info: EtherCAT Master[0]
      DC system time diff. is <= 0 ns`

I guess it's time for real world example now. Read and write examples to enable servo, and move it 10mm.
Any samples for that? I guess it quite depends on specific manufacturer of servo, right ?

Thank you for your help, I'll keep you posted when it moves :)

Apollo3zehn, 2023-05-12:
Congrats :-) In the past I also had some weirdly behaving slaves where a power cycled helped to make them work. My guess is that due to some "playing around" they are in an undefined internal state which should be reset.

I have never controlled a servo, my focus is on measurements. Maybe you need to configure some SDO registers first or you are lucky and the defaults are fine.

If directly controlling the servo using PDOs does not work, you may configure the same project in TwinCAT and look there on the "startup" tab for the correct SDO commands (and then try to understand them using the servo's manual).

[image]

Apollo3zehn, 2023-05-24:
Please follow this example to issue SDO requests during startup:

https://github.com/Apollo3zehn/EtherCAT.NET/blob/8f60d91c220909e030ac77a27edd6411d5abadf2/sample/SampleMaster/Program.cs#L60-L74

ga5tan, 2023-05-24:
that is to add 0x2000 and 0x2001 to be writable?
any idea why Configure may freeze ?

Apollo3zehn, 2023-05-24:
The referenced code enables you to write data to SDO registers during startup. If 0x2000 and 0x2001 are SDO registers (I am not 100% sure) then this would be the appropriate place to manipulate the register contents. It is clear that the index is 0x2000 or 0x2001, respectively, but I do not know what you would use for the _subindex_ parameter for the `SdoWriteRequest(...)` constructor. Example:

```cs
var dataset = new List<object>(); 
dataset.Add((ushort)0x0001); // Operation command = "Execute"
  
var requests = new List<SdoWriteRequest>() 
{ 
    new SdoWriteRequest(0x2001, ??, dataset)    
}; 
```

If 0x2000 is a PDO then you would access it as you wrote above:

```cs
var pdoAnalogIn = slaves[0].DynamicData.Pdos;
var varAnalogIn = pdoAnalogIn[0].Variables.Where(<your variable>).First().DataPtr;
```

According to https://www.ethercat.org/download/documents/EtherCAT_Device_Protocol_Poster.pdf 0x2000 is the start of the manufacturer specific area:

[image]

The `Configure` method is important, but I have never seen it freezing. I am afraid I cannot help you with this.

ga5tan, 2023-05-25:
I found out where it freezes

`EcUtilities.CheckErrorCode(this.Context, EcHL.RequestCommonState(this.Context, (UInt16)SlaveState.Operational), nameof(EcHL.RequestCommonState));`

when I comment it out, app runs further but I get repeated reconfigurations (so I guess something is still amiss)

> Pre-Configure
info: EtherCAT Master[0]
      Found 1 slaves:
      i7DS-**A30 (PDOs: 8 - CSA: 47163)
info: EtherCAT Master[0]
      IO map configured (1 slave, 32 bytes)
info: EtherCAT Master[0]
      DC system time diff. is <= 0 ns
Configured
warn: EtherCAT Master[0]
      state is: 2
warn: EtherCAT Master[0]
      state is: 2
warn: EtherCAT Master[0]
      state is: 2
info: EtherCAT Master[0]
      reconfiguration started
info: EtherCAT Master[0]
      IO map configured (1 slave, 32 bytes)
info: EtherCAT Master[0]
      DC system time diff. is <= 0 ns
info: EtherCAT Master[0]
      reconfiguration successful

Apollo3zehn, 2023-05-25:
> I have no project to configure, neither TwinCAT. If I install TwinCAT will it somehow magically (based on ESI?) know what to do?

ga5tan, 2023-05-25:
I'm sorry, you comment does not seem like it made it trough.
I commented out reconfiguration for now, and will try to read some values
maybe even write

ga5tan, 2023-05-25:
I am still uncertain as to how to read value

`foreach (var pdo in slaves[0].DynamicData.Pdos)
                {

                    foreach (var variable in pdo.Variables)
                    {
                        message.AppendLine($"pdoName '{pdo.Name}' variableName: '{variable.Name}' val: '{variable.DataPtr.ToInt64()}'");
                    }
                }`

This code lists all variables (there seem to be 4 sets of pdo, each with buch of variables. but DataPtr is always 0

I can even find a variable like this:

`var varAnalogIn = pdoAnalogIn[0].Variables.Where(x => x.Name == "Statusword").First();`

but no idea how to get it's value.
Do says DataPtr is pointer, but how do I read value from that pointer?

Apollo3zehn, 2023-05-26:
> I have no project to configure, neither TwinCAT. If I install TwinCAT will it somehow magically (based on ESI?) know what to do?

Yes, TwinCAT knows what to do to get the terminal configured properly. The startup tab will show you which SDO requests are required.

Apollo3zehn, 2023-05-26:
> I'm sorry, you comment does not seem like it made it trough. I commented out reconfiguration for now, and will try to read some values maybe even write

I think this was an accident, I did not write any reply by purpose.

ga5tan, 2023-05-26:
Ha, we commented out lines 47-57 fro  DistributedClocksExtension!
Could that be the problem?
So you  are saying, that the variables are not being synced - copied between servo and local map ?
Yeah, it does not seem as we match the condition on line 185.

I guess SyncManager value -1 is no good? But 2 and 3 could do something?
This is what I have so far:

> Slaves: 1
CreateDynamicData foreach (ln 181)
pdoName: Transmit PDO mapping 1, pdoIndex: 6656, syncManager: 3
pdoName: Transmit PDO mapping 2, pdoIndex: 6657, syncManager: -1
pdoName: Transmit PDO mapping 3, pdoIndex: 6658, syncManager: -1
pdoName: Transmit PDO mapping 4, pdoIndex: 6659, syncManager: -1
CreateDynamicData foreach (ln 181)
pdoName: Receive PDO mapping 1, pdoIndex: 5632, syncManager: 2
pdoName: Receive PDO mapping 2, pdoIndex: 5633, syncManager: -1
pdoName: Receive PDO mapping 3, pdoIndex: 5634, syncManager: -1
pdoName: Receive PDO mapping 4, pdoIndex: 5635, syncManager: -1
syncManagerPdos.Pdo '2' is null!
syncManagerPdos.Pdo '3' is null!
info: EtherCAT Master[0]
      Found 1 slaves:
      i7DS-**A30 (PDOs: 8 - CSA: 47163)
info: EtherCAT Master[0]
      ConfigureIoMap, slave.DynamicData.Pdos: 8
info: EtherCAT Master[0]
      ConfigureIoMap, slave.DynamicData.Pdos: 8
info: EtherCAT Master[0]
      IO map configured (1 slave, 28 bytes)
info: EtherCAT Master[0]
      DC system time diff. is <= 0 ns
Master Configured
info: EtherCAT Master[0]
      pdoName 'Transmit PDO mapping 1' variableName: 'Error code', DataPtr: '0'
      pdoName 'Transmit PDO mapping 1' variableName: 'Statusword', DataPtr: '0'
      pdoName 'Transmit PDO mapping 1' variableName: 'Modes of operation display', DataPtr: '0'
      pdoName 'Transmit PDO mapping 1' variableName: 'Position actual value', DataPtr: '0'
      pdoName 'Transmit PDO mapping 1' variableName: 'Touch probe status', DataPtr: '0'
      pdoName 'Transmit PDO mapping 1' variableName: 'Touch probe pos1 pos value', DataPtr: '0'
      pdoName 'Transmit PDO mapping 1' variableName: 'Following error actual value', DataPtr: '0'
      pdoName 'Transmit PDO mapping 1' variableName: 'Digital inputs', DataPtr: '0'
      pdoName 'Transmit PDO mapping 2' variableName: 'Error code', DataPtr: '0'
      pdoName 'Transmit PDO mapping 2' variableName: 'Statusword', DataPtr: '0'
      pdoName 'Transmit PDO mapping 2' variableName: 'Modes of operation display', DataPtr: '0'
      pdoName 'Transmit PDO mapping 2' variableName: 'Position actual value', DataPtr: '0'
      pdoName 'Transmit PDO mapping 2' variableName: 'Velocity actual value', DataPtr: '0'
      pdoName 'Transmit PDO mapping 2' variableName: 'Torque actual value', DataPtr: '0'
      pdoName 'Transmit PDO mapping 2' variableName: 'Touch probe status', DataPtr: '0'
      pdoName 'Transmit PDO mapping 2' variableName: 'Touch probe pos1 pos value', DataPtr: '0'
      pdoName 'Transmit PDO mapping 2' variableName: 'Digital inputs', DataPtr: '0'
      pdoName 'Transmit PDO mapping 3' variableName: 'Error code', DataPtr: '0'
      pdoName 'Transmit PDO mapping 3' variableName: 'Statusword', DataPtr: '0'
      pdoName 'Transmit PDO mapping 3' variableName: 'Modes of operation display', DataPtr: '0'
      pdoName 'Transmit PDO mapping 3' variableName: 'Position actual value', DataPtr: '0'
      pdoName 'Transmit PDO mapping 3' variableName: 'Velocity actual value', DataPtr: '0'
      pdoName 'Transmit PDO mapping 3' variableName: 'Torque actual value', DataPtr: '0'
      pdoName 'Transmit PDO mapping 3' variableName: 'Touch probe status', DataPtr: '0'
      pdoName 'Transmit PDO mapping 3' variableName: 'Touch probe pos1 pos value', DataPtr: '0'
      pdoName 'Transmit PDO mapping 3' variableName: 'Digital inputs', DataPtr: '0'
      pdoName 'Transmit PDO mapping 4' variableName: 'Error code', DataPtr: '0'
      pdoName 'Transmit PDO mapping 4' variableName: 'Statusword', DataPtr: '0'
      pdoName 'Transmit PDO mapping 4' variableName: 'Modes of operation display', DataPtr: '0'
      pdoName 'Transmit PDO mapping 4' variableName: 'Position actual value', DataPtr: '0'
      pdoName 'Transmit PDO mapping 4' variableName: 'Velocity actual value', DataPtr: '0'
      pdoName 'Transmit PDO mapping 4' variableName: 'Torque actual value', DataPtr: '0'
      pdoName 'Transmit PDO mapping 4' variableName: 'Touch probe status', DataPtr: '0'
      pdoName 'Transmit PDO mapping 4' variableName: 'Touch probe pos1 pos value', DataPtr: '0'
      pdoName 'Transmit PDO mapping 4' variableName: 'Digital inputs', DataPtr: '0'
      pdoName 'Receive PDO mapping 1' variableName: 'Controlword', DataPtr: '0'
      pdoName 'Receive PDO mapping 1' variableName: 'Modes of operation', DataPtr: '0'
      pdoName 'Receive PDO mapping 1' variableName: 'Target position', DataPtr: '0'
      pdoName 'Receive PDO mapping 1' variableName: 'Touch probe function', DataPtr: '0'
      pdoName 'Receive PDO mapping 2' variableName: 'Controlword', DataPtr: '0'
      pdoName 'Receive PDO mapping 2' variableName: 'Modes of operation', DataPtr: '0'
      pdoName 'Receive PDO mapping 2' variableName: 'Target torque', DataPtr: '0'
      pdoName 'Receive PDO mapping 2' variableName: 'Target position', DataPtr: '0'
      pdoName 'Receive PDO mapping 2' variableName: 'Max motor speed', DataPtr: '0'
      pdoName 'Receive PDO mapping 2' variableName: 'Touch probe function', DataPtr: '0'
      pdoName 'Receive PDO mapping 2' variableName: 'Target velocity', DataPtr: '0'
[... cut ...]
      pdoName 'Receive PDO mapping 4' variableName: 'Target torque', DataPtr: '0'
      pdoName 'Receive PDO mapping 4' variableName: 'Max torque', DataPtr: '0'
      pdoName 'Receive PDO mapping 4' variableName: 'Target position', DataPtr: '0'
      pdoName 'Receive PDO mapping 4' variableName: 'Max motor speed', DataPtr: '0'
      pdoName 'Receive PDO mapping 4' variableName: 'Touch probe function', DataPtr: '0'
      pdoName 'Receive PDO mapping 4' variableName: 'Target velocity', DataPtr: '0'

Apollo3zehn, 2023-05-26:
Yes, if you commented out lines 47-57 in the `DistributedClocksExtension` then probably no sync manager is being assigned and that means that no data exchange happens. Sync manager 2 and 3 are for inputs and outputs.

I am wondering why `Transmit PDO mapping 1` and `Receive PDO mapping 1` have sync managers assigned but the DataPtr for variables in these PDOs are still 0. It is difficult for me to says why that happens without debugging the code. I hope you can debug through it and find why DataPtr is not being assigned or is the actual value `0` being assigned to the variables?

Apollo3zehn, 2023-05-26:
It is not required that all PDOs have sync managers assigned. More complex slaves often have different modes of 
operation and only some PDOs are part of the data exchange for the selected mode. But when a sync manager is assigned, then the DataPtr must not be 0. It is important to find out, why they are still 0.

ga5tan, 2023-05-26:
I have uncomented 47-57 but added condition for the loop to go in only when (syncManagerPdos.Pdo != null)
so DataPtr would contain actual value?
How would I find out which PDO contains that 0x2000 register? Or 6040, which seems to be servo enable

ga5tan, 2023-05-26:
Where is the actual assignment of values to DataPrt happening?
Also, even if the mapping is not working, isnt it possible to read write values directly? there is some code in usafe {} directive

ga5tan, 2023-05-26:
I have removed unsafe code block form controller Task loop and now getting this:

> sleepTime: 100
warn: EtherCAT Master[0]
      working counter mismatch 0/3
warn: EtherCAT Master[0]
      working counter mismatch 0/3
warn: EtherCAT Master[0]
      working counter mismatch 0/3
warn: EtherCAT Master[0]
      working counter mismatch 0/3
warn: EtherCAT Master[0]
      working counter mismatch 0/3
warn: EtherCAT Master[0]
      working counter mismatch -1/3
warn: EtherCAT Master[0]
      working counter mismatch 0/3
warn: EtherCAT Master[0]
      working counter mismatch 0/3
warn: EtherCAT Master[0]
      working counter mismatch 0/3
warn: EtherCAT Master[0]
      working counter mismatch 0/3
warn: EtherCAT Master[0]
      working counter mismatch -1/3

that's probably also no good?

Apollo3zehn, 2023-05-26:
> I have uncomented 47-57 but added condition for the loop to go in only when (syncManagerPdos.Pdo != null) so DataPtr would contain actual value? 

I think this extra condition is OK. DataPtr should contain a value, when the sync managers are assigned correctly. If they are still 0 it needs to be debugged. I am sorry, I cannot tell from here, why DataPtr is still 0 for all variables.

> How would I find out which PDO contains that 0x2000 register? Or 6040, which seems to be servo enable
I don't know, honestly. I have never worked with a manufacturer specific register. Maybe the documention tells you, which variable points to register 0x2000. From the log out put you posted before, maybe it is the `Controlword` variable which you can use to enable the servo:

`pdoName 'Receive PDO mapping 1' variableName: 'Controlword', DataPtr: '0'`

From its name I would say this is a variable where you can enable or disable specific bits which sounds as if this is the right spot to enable the servo.

Apollo3zehn, 2023-05-26:
> Where is the actual assignment of values to DataPrt happening? Also, even if the mapping is not working, isnt it possible to read write values directly? there is some code in usafe {} directive

When you create the `EcMaster` instance, it internally allocates a 4096 byte array (the IO map) and stores a pointer to it in the `_ioMapPtr` variable. EtherCAT.NET internally uses [SOEM](https://github.com/OpenEtherCATsociety/SOEM) for the actual communication with the slaves. This library, SOEM, gets the IO map pointer passed in the configuration phase and later SOEM uses that pointer to copy the slave data to the IO map and vice versa.

The variables which are grouped into PDOs point into the IO map. So for example the first variable points to `_ioMapPtr + 0` and the second variable to `_ioMapPtr + 4` (just examples, depends on the variable's data type).

You cannot exchange PDO data with slaves so easily without using SOEM. For that to work the slaves need to be configured properly and then you need to send a special Ethernet/EtherCAT frame over the wire which is read and written by the slaves The returning EtherCAT frame contains the new IO map content. SOEM copies that from the EtherCAT frame to the array where `_ioMapPtr` points to.

Apollo3zehn, 2023-05-26:
> I have removed unsafe code block form controller Task loop and now getting this:

Yes, this is bad. Working counter mismatches mean that the slaves did not respond to the EtherCAT Master's requests.

ga5tan, 2023-05-26:
Yeah, I saw in their Twincat video something about Controlword so I guess it could work, but all the Ptrs are 0
I removed unsecure part from the task, and now running the loop every 3seconds, and checking all Ptrs for non zero after IOread, but they still stay 0 :(

Apollo3zehn, 2023-05-26:
The DataPtr must be nonzero before the data exchange starts. It gets assigned in the configuration phase. In an older post you said you do not have a debugger on the remote machine. 

If you are able to install an OpenSSH server there you can use Visual Studio Code or Visual Studio to remotely debug your application. On Linux this would be easy, on Windows, I don't know. Depends on the Windows version.

If you are not able to remotely debug your problem, I would sprinkle EtherCAT.NET with "Console.WriteLine" statements and print out all relevant info. Especially information about sync manager assignments and DataPtr assignments. 

Or are you using the Nuget version of EtherCAT.NET? Then of course you cannot do what I suggested.

I know this kind of debugging is a pity, and that is one of the reasons why I plan to completely rewrite EtherCAT.NET without the SOEM dependency and make everything in C#. But I don't know when I will find the time for it.

ga5tan, 2023-05-26:
I am installing VS2022 Community on that machine. Wonder how it will run on 4GB or ram.
Yeah, I am using Console.WriteLine (that's some output you saw). It's PITA to copy executables between machines, but it works for now. Currenly dont know where to put  more Console outputs any tips in which functions to add more ?
I have EtherCAT.NET from GIT, so I can see my changes, wonder what happens when you make updates :)

Apollo3zehn, 2023-05-26:
For the start, I would print log messages (or better set a breakpoint if you can debug) here:

https://github.com/Apollo3zehn/EtherCAT.NET/blob/8f60d91c220909e030ac77a27edd6411d5abadf2/src/EtherCAT.NET/EcMaster.cs#L185-L187

The important information here is which variables are being processed in this loop, if any. If you never hit this line, then I guess that the sync mangers are not set correctly. So the next positions to debug would be 

- https://github.com/Apollo3zehn/EtherCAT.NET/blob/8f60d91c220909e030ac77a27edd6411d5abadf2/src/EtherCAT.NET/EcUtilities.cs#L191
- https://github.com/Apollo3zehn/EtherCAT.NET/blob/8f60d91c220909e030ac77a27edd6411d5abadf2/src/EtherCAT.NET/EcUtilities.cs#L214
- https://github.com/Apollo3zehn/EtherCAT.NET/blob/8f60d91c220909e030ac77a27edd6411d5abadf2/src/EtherCAT.NET/Extension/DistributedClocks/DistributedClocksExtension.cs#L55

ga5tan, 2023-05-26:
This is the latest output:
(we are not getting into the loop on line 185 in EcMaster.cs

> Slaves: 1
CreateDynamicData foreach (ln 181)
new SlavePdo osMax0(pdoName: Transmit PDO mapping 1, pdoIndex: 6656, osMax: 0, pdoType.Fixed: False, pdoType.Mandatory: False, syncManager: 3
new SlavePdo osMax0(pdoName: Transmit PDO mapping 2, pdoIndex: 6657, osMax: 0, pdoType.Fixed: False, pdoType.Mandatory: False, syncManager: -1
new SlavePdo osMax0(pdoName: Transmit PDO mapping 3, pdoIndex: 6658, osMax: 0, pdoType.Fixed: False, pdoType.Mandatory: False, syncManager: -1
new SlavePdo osMax0(pdoName: Transmit PDO mapping 4, pdoIndex: 6659, osMax: 0, pdoType.Fixed: False, pdoType.Mandatory: False, syncManager: -1
CreateDynamicData foreach (ln 181)
new SlavePdo osMax0(pdoName: Receive PDO mapping 1, pdoIndex: 5632, osMax: 0, pdoType.Fixed: False, pdoType.Mandatory: False, syncManager: 2
new SlavePdo osMax0(pdoName: Receive PDO mapping 2, pdoIndex: 5633, osMax: 0, pdoType.Fixed: False, pdoType.Mandatory: False, syncManager: -1
new SlavePdo osMax0(pdoName: Receive PDO mapping 3, pdoIndex: 5634, osMax: 0, pdoType.Fixed: False, pdoType.Mandatory: False, syncManager: -1
new SlavePdo osMax0(pdoName: Receive PDO mapping 4, pdoIndex: 5635, osMax: 0, pdoType.Fixed: False, pdoType.Mandatory: False, syncManager: -1
syncManagerPdos.Pdo '2' is null!
syncManagerPdos.Pdo '3' is null!
info: EtherCAT Master[0]
      Found 1 slaves:
      i7DS-**A30 (PDOs: 8 - CSA: 47163)
ConfigureIoMap - before loop, slave.DynamicData.Pdos: 8
ConfigureIoMap - No variables match the criteria on the line 185
ConfigureIoMap - before loop, slave.DynamicData.Pdos: 8
ConfigureIoMap - No variables match the criteria on the line 185
info: EtherCAT Master[0]
      IO map configured (1 slave, 32 bytes)
info: EtherCAT Master[0]
      DC system time diff. is <= 0 ns
Master Configured
waiting for the key...
sleepTime: 3000
info: EtherCAT Master[0]
      No nonzero DataPtrs
info: EtherCAT Master[0]
      No nonzero DataPtrs

ga5tan, 2023-05-27:
I have installed VS 2022 and compiletd EtherCAT.NET from this repo so it can be debugged directly from IDE. If I get remote TeamView do you think you'd find time to have a look?

Apollo3zehn, 2023-05-27:
I would like to reproduce your setup first on my machine. Maybe this is already enough to find out why you get no pointer. I think it should be possible using the ESI file only.

I will do in on Tuesday morning and I will come back to you in any case. If I found no solution we can try to debug remotely (but I would prefer RustDesk instead of Teamviewer).

Apollo3zehn, 2023-05-27:
OK I had some time today and found an issue. The line 
https://github.com/Apollo3zehn/EtherCAT.NET/blob/8f60d91c220909e030ac77a27edd6411d5abadf2/src/EtherCAT.NET/Extension/DistributedClocks/DistributedClocksExtension.cs#L41
resets all previously assigned sync managers to `-1`. This alone is not a problem as normally these would be set again in the following loop:

https://github.com/Apollo3zehn/EtherCAT.NET/blob/8f60d91c220909e030ac77a27edd6411d5abadf2/src/EtherCAT.NET/Extension/DistributedClocks/DistributedClocksExtension.cs#L47-L57

However, the ESI file which you provided is somewhat unexpected as it does not contain any PDOs as written above (https://github.com/Apollo3zehn/EtherCAT.NET/issues/64#issuecomment-1540264540).

So a first workaround would be to comment out this line:
https://github.com/Apollo3zehn/EtherCAT.NET/blob/8f60d91c220909e030ac77a27edd6411d5abadf2/src/EtherCAT.NET/Extension/DistributedClocks/DistributedClocksExtension.cs#L41

I am pretty sure that then you DataPtrs are non-zero.

But I have to find out where the logical error is.

Apollo3zehn, 2023-05-27:
I could not find any description about how to handle situations where the ESI `OpMode` section does not contain any `PDO` elements in the EtherCAT specification. Here is again a screenshot of your ESI file where I would expect one or more `<Pdo>...</Pdo>` entries:

[image]

But from my understanding this means that the sync manager assignment from the `RxPdo` and `TxPdo` sections have to be respected, i.e. EtherCAT.NET should not reset sync managers in this case and let everything as it is. Here is a screenshot which shows the definition of the first `TxPdo` and that this one is assigned to sync manager 3 by default.
[image]

This means, that commenting out the line `_slaveInfo.DynamicData.Pdos.ToList().ForEach(x => x.SyncManager = -1); ` will work in your case. 

I pushed a new commit to master to implement this in a more generic way so that it hopefully works for all slaves (some need the reset, some do not): https://github.com/Apollo3zehn/EtherCAT.NET/commit/f4b939627913c64d75f72999467860522baa9263

If you pull these changes, I think the `DataPtr` is finally non-zero for some variables.

ga5tan, 2023-05-27:
Thank you my friend , I am out of town today, but I appreciate your efforts. Will try your commit ASAP

ga5tan, 2023-06-01:
@Apollo3zehn 
Is the DataPtr containing value itself, or is it just some kind of pointer?
Because I am reading/displaying Statusword like this:

`varAnalogIn = pdoAnalogIn[0].Variables.Where(x => x.Name == "Statusword").First();
                        Console.WriteLine($"Statusword is: {varAnalogIn.DataPtr}");`

But it's different every time I start the app. Assuming the servo did not change status, should be the same. Also just 16bit
I get values like this:

Statusword is: 2400233181067
Statusword is: 1747450662971
Statusword is: 1848892697371

Any idea what could be wrong?

Apollo3zehn, 2023-06-01:
It is a pointer. The sample in this repo shows how you can wrap that pointer in a Span<T> and access the actual value. You can also cast it to a low level C pointer and dereference it. There are different ways in C# to access the actual value. I am on a mobile phone right now, so I cannot provide sample code.

ga5tan, 2023-06-12:
Is this the correct way of accessing 16bit word?
`unsafe
                        {
                            Span<ushort> myStatuswordSpan = new Span<ushort>(varStatusword.DataPtr.ToPointer(), 1);
                            //myStatuswordSpan[0] ^= 1UL << varStatusword.BitOffset;
                            if (Statusword != myStatuswordSpan[0])
                                Console.WriteLine($"Statusword is: {myStatuswordSpan[0]}");
                            Statusword = myStatuswordSpan[0];
}`
Do I need to bit manipulate Span in any way? (like it's commented) ?
Problem is, myStatuswordSpan[0] never changes.

When I read it like this:
`                        unsafe
                        {
                            Span<int> myStatuswordSpan = new Span<int>(varStatusword.DataPtr.ToPointer(), 1);
                            //myVariableSpan[0] ^= 1UL << varStatusword.BitOffset;
                            Console.WriteLine($"Statusword is: {myStatuswordSpan[0]}");
                        }`

it does change, but I guess I am reading more than just 2 bytes, so probably accessing some variable thats stored next to Statusword...

Apollo3zehn, 2023-06-13:
You are accessing the Statusword correctly.

Your second example probably accesses the following:

- `Statusword` (16 bit)
- `Modes of operation display` (8 bit)
- Part of `Position actual value` (32 bit)

[image]

Are you sending commands to the servo? I.e. are you writing data to the `Controlword` variabe? Maybe `Modes of operation` and other output variables must also be written to.

ga5tan, 2023-06-13:
I am not sending any commands yet, but I was hoping Statusword would change when I enable servo via another (Modbus) app designed for commissioning the servo.
So far only 'Position Actual Value' seems to be changing when I jog servo around. And sometimes even when servo is off, which is weird, but it only changes by 1-2. I know that servo always tries to reach specified position and thus always micromoves, but I think it should not, when disabled.
Anyway I may try to write into the Controlword (which is always 0), but wanted to have a feeling that reading is under control

Current output (I only print out if value has changed):
Interesting, Statusword was 34312 whole day yesterday, today is 34368
> sleepTime: 2000
Statusword is: 34368
PosActual is: -73442144
PosActual is: -73442143
PosActual is: -73442144
PosActual is: -73442143
PosActual is: -73442144
PosActual is: -73442143
PosActual is: -73442144
PosActual is: -73442143
PosActual is: -73442144
PosActual is: -73442143
PosActual is: -73442144
PosActual is: -73442143
PosActual is: -73442144

Apollo3zehn, 2023-06-14:
> Every now and then (like 2nd 3rd run of the app) I get this
> 
> > fail: EtherCAT Master[0]
> > SdoWrite failed (0x0000): The working counter is 0 (SOEM).
> 
> is it normal ?

No that should not happen, but that is hard to debug and may be related to the slave itself. Do you have a documentation about the status word? So you know the meaning of each bit?

ga5tan, 2023-06-14:
It seems I have errored out servo. The errors mainly started after I stared modifying Controlword (6,7,Fh)
But I got twice error saying there is no ESI 

> System.Exception: Could not find ESI information of manufacturer '0x88' for slave with product code '0x88710000' and revision '0x0'.

Then I run the app and it's ok again. I guess the closing up of the app may need some more work.
There is variable 'Error code' with Idx: '603F', but I also need to access index '1001h' (Error register) but I dont know how.

ga5tan, 2023-06-14:
This alarm was raised by servo (and I was not able to get out of that error yet)
[image]
Weird thing, isnt it?

ga5tan, 2023-06-14:
Now I am getting quite often this type of error:
EtherCAT message:

Time:1686751739.879 PACKET slave:1 index:1603.00 error:1

ga5tan, 2023-06-15:
Ha, just I poseted of no errors I got firt error of the day:(

> The EtherCAT configuration does not match the actual one. Consider reloading the terminals.

WTH ?

ga5tan, 2023-06-15:
@Apollo3zehn 
CAn you please sdvise how to access this variable?

> <Index>#x6081</Index>
								<Name LcId="1033">Profile velocity</Name>
								<Comment LcId="1033">Profile velocity</Comment>

it's not in the list of variables with DataPtr != 0;
So I dont know how to access it directly. It also probably  isnt synced...
Do you think I could hack ESI and replace some other synced vars or add it ?

[18 more replies]