XzuluX/EtherCAT.NET #65: Bad Image Format Exception

SilentCoyote, 2021-08-19:
Hi,

Having cloned the github project and executed the powershell script as per the instructions for building from source on Windows, when trying to run the sample program I get a BadImageFormatException (System.BadImageFormatException: 'An attempt was made to load a program with an incorrect format. (0x8007000B)') at the call to EcHL.CreateContext() in the ScanDevices method of ECUtilities.

Stack trace:
   at SOEM.PInvoke.EcHL.CreateContext()
   at EtherCAT.NET.EcUtilities.ScanDevices(String interfaceName, SlaveInfo referenceRootSlave) in D:\Projects\VisualStudio\EtherCAT.Net\src\EtherCAT.NET\EcUtilities.cs:line 290
   at SampleMaster.Program.<Main>d__0.MoveNext() in D:\Projects\VisualStudio\EtherCAT.Net\sample\SampleMaster\Program.cs:line 51

Operating system is Windows 10 x64, trying to run Debug, Any CPU build and WinPcap is installed. I've noticed the soem_wrapper.dll in artifacts\bin\SampleMaster\Debug\net50 is reporting 0KB size.

Any help would be appreciated!

Apollo3zehn, 2021-08-19:
I am answering from my phone so I can't check the details, but BadImageFormatException also occurs why trying to load an x64 binary in 32 bit mode or vice versa. But loading an empty file could maybe also raise this exception.

Did compiling the native libraries work (see README)? Did the powershell script finish successfully?

Apollo3zehn, 2021-08-19:
I got an hint that the Nuget package is also not working and containing a 0 kB file. I will check the build process tomorrow morning.

Apollo3zehn, 2021-08-20:
Should be fixed now (nuget package and all branches). I´ll close this for now. If the problem persists, please tell me.

closed 2021-08-20 by Apollo3zehn