XzuluX/EtherCAT.NET #66: can't connect the ethercat

xutingjie168, 2021-09-15:
hello
I can't connect the ethercat,but How to solve the issue？
the funtion ( var rootSlave = EcUtilities.ScanDevices(settings.InterfaceName);)but An error occurred "System.Exception:“The network interface 'eth0' could not be found."  thank you very much.

Apollo3zehn, 2021-09-16:
Maybe it is related to #64. Could you please try the suggested modification and see if it solves your issue?

mgduncan, 2022-01-12:
I am getting the same error - : 'ScanDevices failed (0x0104): One or more terminals did not reach PreOp-state.'. The slave works fine with Beckhoff Master,. Unfortunately, the suggested code changes in Issue 64 are not clear to me - does anyone have modified code for this that addresses this issue?