XzuluX/EtherCAT.NET #67: Is it possible to connect to ELMxxxx EtherCAT measurement terminals?

ToGoOrNotToGo, 2022-04-19:
Hi,

is it possible to connect to ELMxxxx EtherCAT measurement terminals and acquire signal samples with an application?
https://www.beckhoff.com/en-gb/products/i-o/ethercat-terminals/elmxxxx-measurement-technology/

regards

Apollo3zehn, 2022-04-20:
In principle it should be possible, as long as you make the ESI file available to EtherCAT.NET.

ToGoOrNotToGo, 2023-01-24:
Hi, first of all thanks for your answer. It took a while, but now I have an ELM3602-0002 and was able to connect and load all PDOs. However, no samples are transmitted yet. I think it's due to lack of oversampling and IEPE settings.

I am aware of this SOEM issue comment, but it makes no sense (oversampling factor = sync1/syn0), at least when I look into the TwinCAT ELM3602 device configuration:
https://github.com/OpenEtherCATsociety/SOEM/issues/142#issuecomment-361229829

Do you know how the Beckhoff EtherCAT-Oversampling can be configured?

Apollo3zehn, 2023-01-25:
It is long ago that I worked with EtherCAT so I am unfortunately not able to give to a clear solution. As far as I remember and as I can see from the code, there are oversampling settings defined in the ESI file which get loaded here (https://github.com/Apollo3zehn/EtherCAT.NET/blob/master/src/EtherCAT.NET/EcUtilities.cs#L243-L247) and then applied here (https://github.com/Apollo3zehn/EtherCAT.NET/blob/master/src/EtherCAT.NET/EcMaster.cs#L216-L238).

I think as a first step you should debug EtherCAT.NET and see what is happening there. Or you can try to set the `SelectedOpModeId` to the mode you like to have activated:

https://github.com/Apollo3zehn/EtherCAT.NET/blob/master/src/EtherCAT.NET/Extension/DistributedClocks/DistributedClocksExtension.cs#L28

ToGoOrNotToGo, 2023-02-02:
I have now finished the configuration part and tried out a few things. However, I came to the conclusion that it is not possible to run oversampling > 1 stably with normal Windows threading. A constant cycle time of 1ms is simply not possible for Windows, especially since you can only poll (EcMaster.UpdateIO). Is there a way to receive event-based process data in SOEM?

Here is a table for the oversampling commissioning for another ELxxxx:
https://infosys.beckhoff.com/content/1033/el3632/2376439691.html?id=1194465264984357731

Apollo3zehn, 2023-02-02:
You can SpinWait and burn CPU cycles until 1 ms has passed. With Windows I am able to get 10 ms cycle time. But that needs a process with highest priority, thread with highest priority, Windows timer resolution set to 1 ms (https://learn.microsoft.com/en-us/previous-versions/ms713413(v=vs.85)) and an Intel NIC with disabled `interrupt moderation`.

Regarding your event based question: UpdateIO instructs SOEM to send an EtherCAT from to collect process data from the slaves. When that frame returns, the master is updated. There is no other way to get process data. TwinCAT does the same, except that is patched the Windows kernel to become realtime capable for TwinCAT.