XzuluX/EtherCAT.NET #71: soem_wrapper.dll Not found

darklz, 2022-09-22:
I have a liitle problem during build. Its said .....EtherCAT.NET-master\artifacts\bin32\SOEM_wrapper\Release\soem_wrapper.dll" because it was not found. SOEM.PInvoke 

I'm trying to figure out what I'm missing or skip the step to work with library, Can you help me in this point.

what i have done 
install C++ runtime
visual studio 2019

Many thanks in advance.,

[image]

Apollo3zehn, 2022-09-22:
Hi, did you run step 2 of https://github.com/Apollo3zehn/EtherCAT.NET#windows? And was that step successful? It is required to build the native libs.

Apollo3zehn, 2022-09-22:
Step 1 is also necessary, otherwise step 2 will fail

darklz, 2022-09-22:
I have done 1 step., but seem like it dosn't complete. So when i am run second step the Uot put Go like.,

MSBUILD : error MSB1009: Project file does not exist.
Switch: ./artifacts/bin32/SOEM_wrapper/soem_wrapper.vcxproj

Apollo3zehn, 2022-09-22:
Could you please delete the artifacts folder, rerun step 1 and post the full output here?

darklz, 2022-09-23:
I have run step 1 through CMD 

Microsoft Windows [Version 10.0.19043.2006]
(c) Microsoft Corporation. All rights reserved.

C:\Users\ABC\Desktop\EtherCAT.NET-master>pwsh init_solution.ps1
Updating Git submodule.
fatal: not a git repository (or any of the parent directories): .git
Creating native x86 project.

    Directory: C:\Users\ABC\Desktop\EtherCAT.NET-master\artifacts

Mode                 LastWriteTime         Length Name
----                 -------------         ------ ----
d----           9/23/2022  9:01 AM                bin32
-- Selecting Windows SDK version 10.0.18362.0 to target Windows 10.0.19043.
-- The C compiler identification is MSVC 19.25.28612.0
-- The CXX compiler identification is MSVC 19.25.28612.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.25.28610/bin/Hostx64/x86/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.25.28610/bin/Hostx64/x86/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at CMakeLists.txt:11 (add_subdirectory):
  The source directory

    C:/Users/ABC/Desktop/EtherCAT.NET-master/native/SOEM

  does not contain a CMakeLists.txt file.

CMake Deprecation Warning at SOEM_wrapper/virtDev/CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Configuring incomplete, errors occurred!
See also "C:/Users/ABC/Desktop/EtherCAT.NET-master/artifacts/bin32/CMakeFiles/CMakeOutput.log".
Creating native x64 project.
d----           9/23/2022  9:01 AM                bin64
-- Selecting Windows SDK version 10.0.18362.0 to target Windows 10.0.19043.
-- The C compiler identification is MSVC 19.25.28612.0
-- The CXX compiler identification is MSVC 19.25.28612.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.25.28610/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.25.28610/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at CMakeLists.txt:11 (add_subdirectory):
  The source directory

    C:/Users/ABC/Desktop/EtherCAT.NET-master/native/SOEM

  does not contain a CMakeLists.txt file.

CMake Deprecation Warning at SOEM_wrapper/virtDev/CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Configuring incomplete, errors occurred!
See also "C:/Users/ABC/Desktop/EtherCAT.NET-master/artifacts/bin64/CMakeFiles/CMakeOutput.log".

Many Thanks

Apollo3zehn, 2022-09-23:
Thank you, from the log it seems you did download EtherCAT.NET as .zip file instead of `git clone`ing it. It must be a git repository because in step 1 another git submodule will be restored and that fails if you are not in a git repository.

closed 2022-09-23 by darklz