XzuluX/EtherCAT.NET #77: Access violation

ga5tan, 2023-10-31:
Hi

I tried to run your Sample Master as a thread from WinForm app and I am  getting Access Violation Exception

[image]

I run the Master as an async Task before form pops up

[image]

The idea is to run HMI and controller as separate tasks/processes, so the HMI does not freeze...

Any idea what could be the problem?
I know that reading of variables is in unsafe{} block, but it didnt seem that unsafe (meaning it worked), when it was running in your original Sample 

thx
--G

Apollo3zehn, 2023-10-31:
So it is working if you just use the sample master and it fails if you plug it into a Windows Forms app? I don't know what could be the reason for the access violation exception. I would first ensure that the pointer (`varErrorCode.DataPtr`) is non-zero. If that is the case, everything should be fine. EtherCAT.NET allocates the memory for the I/O map itself and access to that map should be safe. Only if the pointer is 0, things should go bad. Or if the pointer calculation is bugged somehow ...

ga5tan, 2023-11-02:
> So it is working if you just use the sample master and it fails if you plug it into a Windows Forms app? 

that is correct.
DataPtr is nonzero:
0.) 'Error code', Idx: '603Fh', DataPtr: '16138129'
0.) 'Statusword', Idx: '6041h', DataPtr: '16138131'
0.) 'Modes of operation display', Idx: '6061h', DataPtr: '16138133'
0.) 'Position actual value', Idx: '6064h', DataPtr: '16138134'
0.) 'Touch probe status', Idx: '60B9h', DataPtr: '16138138'
0.) 'Touch probe pos1 pos value', Idx: '60BAh', DataPtr: '16138140'
0.) 'Following error actual value', Idx: '60F4h', DataPtr: '16138144'
0.) 'Digital inputs', Idx: '60FDh', DataPtr: '16138148'
4.) 'Controlword', Idx: '6040h', DataPtr: '16138120'
4.) 'Modes of operation', Idx: '6060h', DataPtr: '16138122'
4.) 'Target position', Idx: '607Ah', DataPtr: '16138123'
4.) 'Touch probe function', Idx: '60B8h', DataPtr: '16138127'

First comes this exception
Exception thrown: 'System.InvalidOperationException' in mscorlib.dll

and then (often on  _actualWorkingCounter = EcHL.UpdateIo(this.Context, out _dcTime);)
Exception thrown: 'System.AccessViolationException' in EtherCAT.NET.dll
An unhandled exception of type 'System.AccessViolationException' occurred in EtherCAT.NET.dll
Attempted to read or write protected memory. This is often an indication that other memory is corrupt.

ga5tan, 2023-11-12:
And it's not only in SampleMaster where access violation happens. it's also in EcMaster.cs:

[image]

ga5tan, 2023-11-20:
I inserted the code into WPF app, thinking maybe WinForms somehow collides with SOEM dlls, but the problem is the same. Context is null, and UpdateIO fails with access violation.
Anyone reading this, do you have any suggestions as what to try ?

Tmoney2014, 2024-09-06:
I had same ex 
and I did change Thread.sleep -> await Task.Delay()

and It working

I hope, you did find your way to work

rlcobos2, 2025-03-09:
Hi, same problem for me.

Have tryed adding a winforms project in ethercat.net solution.
Have tryed too, in a different PC, creating a new project from zero, and adding the libraries from nuget.

same result.

I'm using  .NET 8.0.

[image]

Any suggestion??