XzuluX/EtherCAT.NET #78: Platform detection for init_solution.ps1 - to handle PlatformNotSupportedException

joachim-heck, 2024-08-02:
Below i have an updated version which adds an automatic platform detection to init_solution - maybe somebody finds this helpful when struggeling with not supported platform exception during execution.

```
# Define the platform
$IsWindows = $false
$IsLinux = $false

# Check the OS platform
if ($env:OS -eq "Windows_NT") {
    $IsWindows = $true
} elseif ($env:OSTYPE -eq "linux-gnu") {
    $IsLinux = $true
} else {
    throw [System.PlatformNotSupportedException] "Unsupported OS."
}

Write-Host "Updating Git submodule."
git submodule update --init --recursive --quiet

# x86
Write-Host "Creating native x86 project."
$path = "$($PSScriptRoot)/artifacts/bin32"
New-Item -Force -ItemType directory -Path $path
Set-Location -Path $path

if ($IsWindows) {
    cmake ./../../native -DCMAKE_CONFIGURATION_TYPES:STRING="Debug;Release" -G "Visual Studio 16 2019" -A "Win32"
} elseif ($IsLinux) {
    cmake ./../../native -DCMAKE_BUILD_TYPE=Release -DCMAKE_C_FLAGS=-m32 -DCMAKE_CXX_FLAGS=-m32 -DCMAKE_TRY_COMPILE_TARGET_TYPE=STATIC_LIBRARY
} else {
    throw [System.PlatformNotSupportedException]
}

# x64
Write-Host "Creating native x64 project."
$path = "$($PSScriptRoot)/artifacts/bin64"
New-Item -Force -ItemType directory -Path $path
Set-Location -Path $path

if ($IsWindows) {
    cmake ./../../native -DCMAKE_CONFIGURATION_TYPES:STRING="Debug;Release" -G "Visual Studio 16 2019" -A "x64"
} elseif ($IsLinux) {
    cmake ./../../native -DCMAKE_BUILD_TYPE=Release -DCMAKE_C_FLAGS=-m64 -DCMAKE_CXX_FLAGS=-m64
} else {
    throw [System.PlatformNotSupportedException]
}

# return
Set-Location -Path $PSScriptRoot

```

Apollo3zehn, 2024-08-02:
Thanks :-) `$IsWindows` and `$IsLinux` are built-in Powershell variables so there should be no need to redefine them. Which Powershell version do you use?

joachim-heck, 2024-08-03:
Ough,...my fault ;-D
Maybe this is due to I'm using the default PowerShell Desktop from Windows 11 install.

```
Name                           Value
----                           -----
PSVersion                      5.1.22621.3880
PSEdition                      Desktop
PSCompatibleVersions           {1.0, 2.0, 3.0, 4.0...}
BuildVersion                   10.0.22621.3880
CLRVersion                     4.0.30319.42000
WSManStackVersion              3.0
PSRemotingProtocolVersion      2.3
SerializationVersion           1.1.0.1
```

Apollo3zehn, 2024-08-05:
Ah okay, yes this is a very old Powershell version :-) I will close this issue for now.

closed 2024-08-05 by Apollo3zehn