XzuluX/EtherCAT.NET #85: How to read SDO variables?  是我的使用方式的问题吗？

afate2020, 2024-12-03:
After using the fix idea of #64, why does the sample code like sampleTest use EcUtilities.SdoRead to read the data, and the reading output is 0?

Apollo3zehn, 2024-12-03:
That looks good to me, did you try other SDO indices as well?

afate2020, 2024-12-04:
By using the avalonia ui, I traversed all the SDO indexes of the esi and the results were all 0, I referred to the internal code, checked for errors, confirmed al-state and act-state, the error codes were both 0x0000, the status was OP (check confirmation via wireshark), here are my relevant code , wireshark view results and avalonia ui results.
```csharp
public void UpdateValue()
{
    if (bitSize <= 32)
    {
        var baseValue = new byte[4];
        int count = EcUtilities.SdoRead(_parent.Master.Context, _parent.slaveIndex, Index, subIndex, ref baseValue);
        try
        {
            EcUtilities.CheckErrorCode(_parent.Master.Context, count, nameof(EcUtilities.SdoRead));
        }
        catch (Exception ex)
        {
            string description = EcUtilities.GetSlaveStateDescription(_parent.Master.Context, [_parent.SlaveInfo]);
            Debug.WriteLine(ex.Message);
            Debug.WriteLine(description);
        }
        finally
        {
            Debug.WriteLine($"Read SDO: {BitConverter.ToInt32(baseValue, 0)},backCount: {count}");
        }
        if (baseValue == _value)
        {
            _value = baseValue;
        }
    }
}
```
[image]
[image]

> Read SDO: 0,backCount: 0
SdoRead failed (0x0000): The working counter is 0 (SOEM).
Slave   1 | CA: False | Req-State: 0x0008 | Act-State: 0x0008 | AL-Status: 0x0000 | Sys-Time Diff: 0x00000000 | Speed Counter Diff: 0x0000 | #Pdo out: 26 | #Pdo in: 22 | (cspcsv)

Apollo3zehn, 2024-12-04:
I think for SDO to work you need to be in PreOP state as stated in the thread you linked in the first post.

afate2020, 2024-12-04:
> I think for SDO to work you need to be in PreOP state as stated in the thread you linked in the first post.

In my avalonia ui,if I select network card,it will start master to OP state and create thread Task like sampleTest,the wireshark report can check it.The OP state can also read and write SDO.

In SampleTest Code, I tested another index, and that's right.In first post,this is my mistake, I got the subIndex wrong.

Now In my avalonia ui, I found my slaveIndex is wrong.By fix it,the state of the app is as I expected.
That's also my mistake. ><

I want to create ethercat Avalonia app universal version in github,what need I to do and look out for？I implemented the persistence of the selected network card and the corresponding tree grid of tab selection,by create new model from your base xml esi model.

closed 2024-12-05 by afate2020

Apollo3zehn, 2024-12-06:
> I want to create ethercat ui universal version in github,what need I to do and look out for

Sounds interesting :-) I would install TwinCAT and see what they do with their UI so you know what is minimally required

afate2020, 2024-12-13:
> > I want to create ethercat ui universal version in github,what need I to do and look out for
> 
> Sounds interesting :-) I would install TwinCAT and see what they do with their UI so you know what is minimally required

Thanks your help!