XzuluX/EtherCAT.NET #88: msbuild ./artifacts/bin32/SOEM_wrapper/soem_wrapper.vcxproj /p:Configuration=Release 执行这个的时候总是错误

SimpleZDM, 2025-07-25:
LNK2015无法解析的外部符号 imp timeEndPeriod，函数 ec closenic 中引用了该符号
LNK201无法解析的外部符号imp timeBeginPeriod，函数 osal getrelativetime 中引用了该符号LNK2015无法解析的外部符号imp ntohs，函数 ecx inframe 中引用了该符号
LNK2001无法解析的外部符号imp ntohs
LNK2015无法解析的外部符号imp htons,函数 ec setupheader 中引用了该符号LNK2001无法解析的外部符号 imp htons
LNK201(无法解析的外部符号 pcap sendpacket，函数 ec outframe 中引用了该符号LNK201(无法解析的外部符号 pcap open，函数 ecx setupnic 中引用了该符号LNK201无法解析的外部符号 pcap next ex，函数 ecx inframe 中引用了该符号LNK2015无法解析的外部符号 pcap freealldevs，函数 oshw find adapters 中引用了该符号LNK201!无法解析的外部符号 pcap findalldevs，函数 oshw find adapters 中引用了该符号

Apollo3zehn, 2025-07-25:
Thanks for your issue. The repository language is English, so please translate your Problem to English. Thanks

SimpleZDM, 2025-07-28:
in windows 11  environment,use visual studio 2022,execute msbuild ./artifacts/bin32/SOEM_wrapper/soem_wrapper.vcxproj /p:Configuration=Release, There is a compilation error. 
What should be done?

SimpleZDM, 2025-07-28:
Sorry, I tried a few more times and it's fine now